openwrt/luci #3121: ntp client with only "Use DHCP advertised servers"

mistepien, 2019-09-27:
Now even if you turn on option "Use DHCP advertised servers" in ntp client configuration, you have to add servers to the list (actually they are added automatically and cannot be removed).

feckert, 2019-09-27:
Could you please add more information? What page?

mistepien, 2019-09-27:
System -> System -> Time Synchronization

And what I ask for (ntpclient with empty server list using only ntpserver given by DHCP) is possible
according to uci:

> system.ntp=timeserver
ucitrack.@ntpclient[0]=ntpclient
ucitrack.@ntpclient[0].init='ntpclient'

I this case in luci I have:
Enable NTP client: yes
Provide NTP server: no
Use DHCP advertised servers: yes
empty server list

In system such process is running:
> /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 192.168.1.8

192.168.1.8 is address of machine with ntpserver given by dhcp in that case.

The point that to get to this point I had to edit /etc/config/system by myself.

commit ad05b61d (2019-09-27): luci-mod-system: allow deleting all NTP servers

Fixes: #3121
closed 2019-09-27 by jow-