openwrt/luci #3123: luci-base: does not show encryption in mesh mode

castillofrancodamian, 2019-09-27:
[image]

[image]

jow-, 2019-09-28:
Please provide the following outputs:

```
opkg list-installed | grep iwinfo
opkg list-installed | grep luci-base
cat /var/run/wpa_supplicant*.conf
cat /var/run/hostapd*.conf
iw list

castillofrancodamian, 2019-09-28:
```
root@NOBLEX-NT1013E:~# opkg list-installed | grep iwinfo
iwinfo - 2019-09-22-313e8270-1
libiwinfo-lua - 2019-09-22-313e8270-1
libiwinfo20181126 - 2019-09-22-313e8270-1
rpcd-mod-iwinfo - 2019-09-21-95f0973c-1
```
```
root@NOBLEX-NT1013E:~# opkg list-installed | grep luci-base
luci-base - git-19.270.39227-355223e-1
```
```
root@NOBLEX-NT1013E:~# cat /var/run/wpa_supplicant*.conf

country=AR
network={

        ssid="TEST"
        key_mgmt=SAE
        mode=5
        mesh_fwding=1
        mesh_rssi_threshold=0
        fixed_freq=1
        frequency=2412
        ht40=1
        max_oper_chwidth=0
        sae_password="00000000"
        beacon_int=100
}
```
```
root@NOBLEX-NT1013E:~# iw list
Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        RTS threshold: 2347
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x186e
                        HT20/HT40
[... 88 lines cut ...]
        software interface modes (can always be added):
                 * AP/VLAN
                 * monitor
        interface combinations are not supported
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports TX status socket option.
        Device supports HT-IBSS.
        Device supports SAE with AUTHENTICATE command
        Device supports low priority scan.
        Device supports scan flush.
        Device supports AP scan.
        Device supports per-vif TX power setting
        Driver supports full state transitions for AP/GO clients
        Driver supports a userspace MPM
        Device supports configuring vdev MAC-addr on create.
        Supported extended features:
                * [ RRM ]: RRM
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
```

jow-, 2019-09-30:
Please install `wpa-cli`, and provide the output of `wpa_cli -p /var/run/wpa_supplicant -i wlan0 status`

jow-, 2019-09-30:
Fixed with https://github.com/openwrt/openwrt/commit/f2b9181bb1ba9f0a3a88b8c47754d05efe93298a

closed 2019-09-30 by jow-