openwrt/luci #3124: Question: should i port ddns?

Ansuel, 2019-09-28:
@jow- sorry for the random issue... i notice you are converting all the model to client side implementation... should the app be converted too ?

jow-, 2019-09-30:
@Ansuel - if you feel like it then yes, it would be a good idea. But its not a strict requirement yet.

feckert, 2019-10-21:
@Ansuel Are you still planning to change the app to the new Javascript?

Ansuel, 2019-10-21:
Mh yes but now I'm working on something else.
I need to study the new API to convert all to js... Also it would be good
to "clean" the module...
Why the question?

feckert, 2019-10-21:
I have come up with a few things that I did not like so much with this app. If you rewrite this app we could fix this.
* Link to the openwrt page (should be fixed in a general way)
* General page should be a tab and not a link
* And start/stop should be done on save if the section is enabled

Ansuel, 2019-10-21:
Yes I notice that too...

There is also some strange thing with general ddns option hided in a link
instead of putting them in a tab...

Ansuel, 2019-10-28:
Anyway i have started the porting of the ddns app... For now i have ported the index page and the global config options

Ansuel, 2019-11-06:
@feckert https://github.com/openwrt/luci/pull/3267 almost done... i think i just need to implement custom script implementation... I need some hint about the name of the js (overview.js is a bit misleading) and a general review of the code and of the functions... I removed some checks and it would be good to have a second opinion if the extra checks should be reintroduced.

feckert, 2019-11-06:
@Ansuel see my comments in #3267

closed 2019-11-16 by Ansuel