openwrt/luci #3129: How can I set the ipt_args option, thank you

vbzc, 2019-09-30:
I want to use this function to set the port of the proxy. I tried a lot of writing methods and it is not successful. Please tell me how to set it in luci, thank you.

jow-, 2019-09-30:
There is no `ipt_args` function in LuCI.

closed 2019-09-30 by jow-

vbzc, 2019-10-01:
This is the setting file content in shadowsocks, which has option ipt_args
vi /etc/config/shadowsocks-libev

config ss_rules 'ss_rules'
        option redir_tcp 'hi'
        option src_default 'checkdst'
        option dst_ips_bypass_file '/etc/chinadns_chnroute.txt'
        option dst_default 'forward'
        option local_default 'checkdst'
        option redir_udp 'hi'
        list src_ips_bypass '10.0.1.26'
        list src_ips_bypass '192.168.1.104'
        list ifnames 'br-lan'
        option ipt_args '--dport 21:1024'

vbzc, 2019-10-01:
This is a screenshot of the luci settings interface

https://1drv.ms/u/s!AiC3DiZkYVCHhNAxalebwY-UPfYAUg