openwrt/luci #3136: luci-mod-system: firstboot option is not displayed

musashino205, 2019-10-01:
- device:
  I-O DATA WN-AC1600DGR (16 MiB SPI-NOR flash)

```
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00e50000 00010000 "firmware"
mtd3: 001c0000 00010000 "kernel"
mtd4: 00c90000 00010000 "rootfs"
mtd5: 00520000 00010000 "rootfs_data"
mtd6: 00100000 00010000 "manufacture"
mtd7: 00010000 00010000 "backup"
mtd8: 00050000 00010000 "storage"
mtd9: 00010000 00010000 "art"
```

- browser:
  - Microsoft Edge 44.18362.329.0
  - Google Chrome 69.0.3497.100（Official Build）
  - Mozilla Firefox 69.0

"Reset to defaults" option is not displayed in admin/system/flash page. It looks like getting the partition layout has failed.

jow-, 2019-10-02:
What is reported by `cat /sys/devices/virtual/mtd/mtd5/name` ?

1Jeff1, 2019-10-02:
It look like not all devices are affected on ipq40xx  netgear EX6150v2 is not displayed
[image]
 
On Linksys WRT1900ACS is normal
[image]

jow-, 2019-10-02:
@1Jeff1 , can you check `cat /sys/devices/virtual/mtd/mtd*/name` on the affected device, too?

1Jeff1, 2019-10-02:
root@OpenWrt:~# cat /sys/devices/virtual/mtd/mtd*/name 
cat: can't open '/sys/devices/virtual/mtd/mtd*/name': No such file or directory
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "SBL1"
mtd1: 00020000 00010000 "MIBIB"
mtd2: 00060000 00010000 "QSEE"
mtd3: 00010000 00010000 "CDT"
mtd4: 00010000 00010000 "DDRPARAMS"
mtd5: 00010000 00010000 "APPSBLENV"
mtd6: 00080000 00010000 "APPSBL"
mtd7: 00010000 00010000 "ART"
mtd8: 00010000 00010000 "config"
mtd9: 00010000 00010000 "pot"
mtd10: 00010000 00010000 "dnidata"
mtd11: 00e10000 00010000 "firmware"
mtd12: 00340000 00010000 "kernel"
mtd13: 00ad0000 00010000 "rootfs"
mtd14: 001f0000 00010000 "rootfs_data"
mtd15: 00040000 00010000 "language"

jow-, 2019-10-02:
Thank you - wiIll rework the logic.

commit 020ac1e0 (2019-10-02): luci-mod-system: flash.js: fix "rootfs_data" partition detection

The `/sys/devices/virtual/mtd/` information is not available on all
devices so rely on the contents of `/proc/mtd` instead to check for
the existence of the `rootfs_data` partition.

Fixes: #3136
closed 2019-10-02 by jow-