openwrt/luci #3140: luci-nginx: error flashing new firmware with web interface

jammmet, 2019-10-02:
Trying to update firmware fails during upload using nginx as the LuCI web server.
Console output attached:

[image]

jammmet, 2019-10-02:
That was on Safari.
On Firefox, a different error manifests:

[image]

And on Chrome:

[image]

jow-, 2019-10-02:
This is a bug in the `luci-nginx` package, most likely because it is not configured to execute CGI programs correctly. 

@Ansuel

jow-, 2019-10-02:
Please follow up on https://github.com/openwrt/packages/issues/10134 - this is not a LuCI bug.

closed 2019-10-02 by jow-