openwrt/luci #3143: luci-app-aria2 Serious Error!

zouyujia, 2019-10-04:
1./etc/init.d/aria2     line:238
	          elif [ "$rpc_auth_method" = "user_pass" ]; then
line238	         	if [ -n "$rcp_user" ]; then
			      append_setting "rpc-user=${rpc_user}"
			append_setting "rcp-passwd=${rcp-passwd}"
2.\usr\lib\lua\luci\model\cbi\aria2.lua     line:78
                   o = s:taboption("general", Value, "rpc_user", translate("RPC username"))
                   o:depends("rpc_auth_method", "user_pass")
line:78         o.rmempty = false

feckert, 2019-10-07:
Which branch/version off OpenWrt do you use?

zouyujia, 2019-10-07:
> Which branch/version off OpenWrt do you use?

luci-app-aria2 1.0.1-2
openwrt-18.06 branch (git-19.020.41695-6f6641d)

zouyujia, 2019-10-07:
releases 18.06.2

dibdot, 2019-10-07:
That's an error in the base package and not in the luci app, which has been already fixed in master (see here for reference: https://github.com/openwrt/packages/commit/ed9514e93e16ad474497fc02071956966de2cf2c#diff-c0d268204a9d5cf9f053256ea9aad529)

closed 2019-10-07 by dibdot

zouyujia, 2019-10-08:
This version is not fixed.
http://downloads.openwrt.org/releases/18.06.2/packages/arm_cortex-a9/packages/aria2_1.33.0-1_arm_cortex-a9.ipk

Date:Sun Oct 6 20:48:01 2019

dibdot, 2019-10-08:
If switching to master is no option for you, then open a ticket at the package repo and ask the aria2 maintainer, if he is willing to backport the fix to 18.06 branch.