openwrt/luci #3147: luci-base: form.js titleFn return null when attr is not defined

ysc3839, 2019-10-07:
https://github.com/openwrt/luci/blob/90c52a1782ae7727c0a430dcec64d9c1b072e452/modules/luci-base/htdocs/luci-static/resources/form.js#L229-L244
When `attr` is not defined, `s` remains `null`, then it returns `null`, which result in null name in `TableSection`.
[image]
@jow-

jow-, 2019-10-07:
@ysc3839 - the `titleFn()` is supposed to return `null` so this part is fine, the table is not supposed to be rendered with `null` section titles though. Can you provide the exact example .js code you used?

ysc3839, 2019-10-07:
@jow- Just a TableSection. I found here titleFn returns null, resulting null name in table.
https://github.com/openwrt/luci/blob/90c52a1782ae7727c0a430dcec64d9c1b072e452/modules/luci-base/htdocs/luci-static/resources/form.js#L1032
So I guess it mistakenly return null?

jow-, 2019-10-07:
Let me check...

commit e7ca163d (2019-10-07): luci-base: form.js: fix anonymous section table titles

When no section title is defined, the table renderer is supposed to use
the uci section id as fallback when the table section is not declared
to be anonymous.

Fixes: #3147
Fixes: a90bf384b ("luci-base: form.js: harmonize title property handling")
closed 2019-10-07 by jow-