openwrt/luci #3149: luci-theme-material: missing panel title in with client-side view

ysc3839, 2019-10-08:
When using client-side view, panel title is missing:
[image]
Server-side view is normal:
[image]

I think it's because the code that creates it excutes on load, but the view is not loaded at that time.
https://github.com/openwrt/luci/blob/e459683822925bcdeb552673a4811aa3b4322640/themes/luci-theme-material/htdocs/luci-static/material/js/script.js#L174-L180

jow-, 2019-10-09:
Yet another material theme specific bug.

jow-, 2019-12-16:
Fixed as side-effect of 92eecedc8a52b0ba814daddb5167449a2ebb7f26

ysc3839, 2020-01-18:
This issue still not fixed.
[image]

Version is LuCI (git-20.016.76613-0608ff0) / OpenWrt SNAPSHOT r12034-1322190fd3.

Activate a breakpoint in script.js, at this time `legend` elements is not loaded.
[image]

jow-, 2020-01-18:
Okay, someone who cares about material needs to look into this then.

solidus1983, 2020-01-28:
there is quite a few bugs in the Material theme, however it only shows in master (19.07 rc2 branch is fine). So yes it would be down to the latest changes made.

I was looking into it for my theme which is based on Material however my only system decided to take a dump of death.

oldium, 2020-10-11:
I am on current master branch and can confirm this is still there. The trigger `luci-loaded` is executed when actually no `legend` element exists yet. The GUI is created asynchronously, so the conversion from `legend` to `span` needs to take place some time later. Any chance to get this fixed?

oldium, 2020-10-12:
The `View` class will most probably need to emit a notification when the DOM is ready - in `luci.js` in constructor of `View` class:

https://github.com/openwrt/luci/blob/0d0816ca107c84631ab03b5ef2e6d373e102d09a/modules/luci-base/htdocs/luci-static/resources/luci.js#L1945-L1958

jow-, 2020-10-12:
I have no interest in adding additional logic to fix the material theme, the theme should be fixed no not fiddle with the DOM. Where it makes sense we could think about revising the default markup to accommodate for more complex CSS.

oldium, 2020-10-12:
The `legend` element should be first child of `fieldset` element, but the code does not create any `fieldset`, so it looks like the Material theme is trying to fix the HTML structure actually.

jow-, 2020-10-12:
Long time ago, section elements used to be `<fieldset>`, nowadays they're `<div>` ones but the legend tag remained as a leftover. I did some cursory googling and it appears `<legend>` is indeed only legal below `<fieldset>` and quirky to style in some situations, which likely is the reason why material tries to replace it.

So `<legend>` should be changed to `<div>` or `<span>` in `form.js` and the legacy templates then.

oldium, 2020-10-12:
Something like this:

Original:
https://github.com/openwrt/luci/blob/0d0816ca107c84631ab03b5ef2e6d373e102d09a/modules/luci-base/htdocs/luci-static/resources/form.js#L3140-L3141

Changed:
```javascript
 if (typeof(this.title) === 'string' && this.title !== '') 
 	sectionEl.appendChild(E('div', { 'class': 'cbi-section-title' }, this.title));
```
```

oldium, 2020-10-12:
Other places are trying to render `this.title` as `<h2>` or `<h3>`, which would make sense too.

jow-, 2020-10-12:
Right, the standard `openwrt`, `bootstrap` and `openwrt2020` themes treat legend equivalent to an h3, so we should use that instead and fix whatever CSS breaks due to this.

oldium, 2020-10-12:
The Material theme is doing some other DOM tweaks, so the question is whether it actually does anything - especially when views are being loaded asynchronously, so they do not exist when those tweaks are tried to be made.

oldium, 2020-10-12:
Material theme is not the only one, also Rosy theme is trying to manipulate DOM. Both themes need to be updated to work with client-side rendered views, maybe rendering code of `form.js` could be updated _somehow_ to allow changes by themes.

jow-, 2020-10-12:
Well, I don't particularly like either theme. Rosy went out of maintenance the moment it got dropped into the tree and Material has these pesky jQuery dependency and ugly DOM manipulations (which Rosy copy-pasted afair).

I prefer to fix the generic markup to be better workable with CSS styling instead of allowing themes to manipulate the DOM. I think that just leads to maintenance hell and incompatibilities.

PR #4538 (2020-10-30): luci-base/luci-compat: Replace legend element with h3.
The `legend` HTML element is allowed only as first element in `fieldset`, so use `h3` instead, which is rendered the same within themes.

Fixes #3149.
commit 4775d1ad (2020-10-30): luci-compat: Replace legend element with h3.

The legend HTML element is allowed only as first element in fieldset, so
use h3 instead, which is rendered the same within themes.

Fixed #3149.
commit ebaa50ec (2020-10-30): luci-base: Replace legend element with h3.

The legend HTML element is allowed only as first element in fieldset, so
use h3 instead, which is rendered the same within themes.

Fixes #3149.
closed 2020-10-30 by jow-