openwrt/luci #3150: luci-base: strange section tabs with client-side view after login

ysc3839, 2019-10-08:
#### Reproduce steps:
1. Logout
2. Go to http://192.168.1.1/cgi-bin/luci/admin/services/test
3. Login
4. Result: 
[image]

This bug is not always reproducible, and it seems that it only occur after login. It doesn't occur by reloading the page. My browser is Google Chrome Version 77.0.3865.90.
I found it's caused by this code:
https://github.com/openwrt/luci/blob/e459683822925bcdeb552673a4811aa3b4322640/modules/luci-base/htdocs/luci-static/resources/ui.js#L2016-L2045
Sometimes the view is loded before this code execute, so that `document.querySelectorAll('[data-tab]')` selects the tabs.
I guess it's some kind of race condition. There shoule be some restriction to ensure the view is loaded after that.

#### Code:
/usr/lib/lua/luci/controller/test.lua
```
module("luci.controller.test", package.seeall)

function index()
	entry({"admin", "services", "test"}, view("test"), _("test")).dependent = true
end
```
/etc/config/test
```
config test1
	option opt1 'opt1'

config conf common
	option opt2 'opt2'

config conf conf1
	option opt3 'opt3'
```
/www/luci-static/resources/view/test.js
```
'use strict';
'require form';

return L.view.extend({
	render: function() {
		var m, s, o;

		m = new form.Map('test', 'test');

		s = m.section(form.TypedSection, 'test1', 'section');
		s.anonymous = true;

		s.tab('tab1', _('tab1'));
		s.tab('tab2', _('tab2'));

		for (var i = 0; i < 20; ++i) {
			s.taboption('tab1', form.Value, 'opt' + i, 'Opt' + i);
		}

		o = s.taboption('tab2', form.SectionValue, 'common', form.NamedSection, 'common', 'conf', 'tab2');
		s = o.subsection;

		s.option(form.Flag, 'opt', 'opt');

		return m.render();
	}
});
```

closed 2019-10-22 by jow-

jow-, 2019-10-22:
It was quite hard to debug, whenever I added some debugging code, the race condition stopped occurring but now I'm confident that it is solved.

ysc3839, 2019-10-22:
@jow- Thanks for solving!

commit 2b7ca791 (2019-10-22): luci-base: ui.js: prevent race condition in tab initialization

Fixes: #3150