openwrt/luci #3156: broken save mtd function

Ansuel, 2019-10-08:
I notice that save mtd function is broken since it does report Invalid characters 

I run this with uhttpd, tell me if it's just a problem on my side.

jow-, 2019-10-09:
Please provide more details. What is reporting invalid characters where?

Ansuel, 2019-10-09:
As soon as I click  save mtdblock button in flash page

(It's from cgi-download )

hnyman, 2019-10-09:
Works for me with ipq806x/R7800 and ath79/WNDR3700v2

LuCI Master (git-19.280.64531-5ccbda5) / OpenWrt SNAPSHOT r11167-273a6cb562

jow-, 2019-10-09:
Sounds as if your browser is displaying the binary contents inline and not offering them as download. I cannot reproduce this here either using Chrome or Firefox.

closed 2019-10-09 by jow-

Ansuel, 2019-10-09:
@jow- pls reopen... i modified cgi-io to include some debug output....

Wed Oct  9 18:55:04 2019 daemon.info uwsgi: No-Lag-Router.mtd0.bin
Wed Oct  9 18:55:04 2019 daemon.info uwsgi: Invalid -

By renaming my hostname to "no" 
it does download the file

Ansuel, 2019-10-09:
```
	for (p = fields[5]; p && *p; p++)
		if (!isalnum(*p) && !strchr(" ()<>@,;:[]?.=%", *p)) {
			fprintf(stderr, "Invalid %c\n", *p);
			return failure(400, 0, "Invalid characters in filename");}

	for (p = fields[7]; p && *p; p++)
		if (!isalnum(*p) && !strchr(" .;=/-", *p))
			return failure(400, 0, "Invalid characters in mimetype");
```

change isalnum to isgraph ?

Ansuel, 2019-10-09:
Here the file pr 
https://github.com/openwrt/packages/pull/10191