openwrt/luci #3161: luci-app-mosquitto

finasfinas, 2019-10-09:
Can't find it ( current trunk ) on the application list.
I have luci admin full selected.

jow-, 2019-10-09:
There is no such application in the repos.

closed 2019-10-09 by jow-

feckert, 2019-10-09:
https://github.com/openwrt/luci/tree/master/applications/luci-app-mosquitto ?

jow-, 2019-10-09:
Didn't know that this is a thing. Anyway, the original report is far too vague to act upon. My guess is that mosquitto failed to build for some random arch and therefore luci-app-mosquitto isn't available. Either way not a bug we can solve here.

finasfinas, 2019-10-09:
@jow- 
Sorry for the vagueness. 

What I mean is that when I do a make menuconfig, select luci, and them go to the applications list, I cannot find there the luci-app-mosquitto.
I saw on the code that it depends on luci admin full and on mosquitto and even having those enabled does not make it appear on the list.

finasfinas, 2019-10-09:
So,
Selecting the package mosquitto-nossl or mosquitto-ssl will make luci-app-mosquitto appear.
Doesn't seem to be the standard way of doing things, namely, all packages are viewable.