openwrt/luci #3162: luci-app-simple-adblock: DIV error with version git-19.275.23087-48f5..-37

josephreynolds1, 2019-10-09:
Maintainer: @\Stan Grishin
Environment: (arm_cortex-a15_neon-vfpv4 , Netgear r7800, 18.06.4 r7808-ef686b7292)

Description: luci-app-simple-adblock fails to load with a DIV error since version git-19.275.23087-48f5..-37.

XML Parsing Error: mismatched tag. Expected: .
Location: https://192.168.1.1:444/cgi-bin/luci/admin/services/simple-adblock
Line Number 179, Column 232:

hnyman, 2019-10-09:
cc @stangri

stangri, 2019-10-10:
Can't reproduce.

dibdot, 2019-10-10:
@stangri Use the XHTML-based OpenWrt standard theme to reproduce this issue ... at least the used br-tag syntax is not "well formed" - use `<br />` instead (if you really need hard text breaks).. I haven't tested this app, maybe there are more problems like that ...

closed 2019-10-10 by feckert