openwrt/luci #3177: username and password different from factory

rctito55, 2019-10-13:
after resetting the mini router, when writing user: root and password: admin I get "Invalid username and / or password! Please try again." Could someone help me? This forum may not be for that but I can't find a way out of the problem. Thanks in advance

ysc3839, 2019-10-13:
By default root password is empty. Just click the login button.
Or are you using some modded firmware?

rctito55, 2019-10-13:
It is likely that the firmware was modified because, after factory reset, it comes out in ip 192.168.8.1, instead of 192.168.1.1 as in the user manual. Try these user and pass and it is none:
admin admin, root root, 1234 1234, its combinations and also root and without password and it is not ... any suggestion is welcome

rctito55, 2019-10-13:
Only the default user that is root is maintained, as a suggestion, in the textbox. If there is any way to return the factory firmware it would be very important to know

hnyman, 2019-10-13:
> after factory reset, it comes out in ip 192.168.8.1, instead of 192.168.1.1 as in the user manual. Try these user and pass and it is none: admin admin, root root, 1234 1234, its combinations and also root and without password and it is not ... any suggestion is welcome

You might try using the failsafe mode. It can be triggered in the early boot process and it enables you to enter the ssh console without password. (just the username root, if asked)
https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset#failsafe_mode 

(note that entering the failsafe mode requires you to press a button on the router during a specific two-second time window during the boot process. The exact moment depends on the router.

But as your issue has nothing to do with LuCI GUI and is not a bug, I close this issue.

closed 2019-10-13 by hnyman