openwrt/luci #3182: Wireless edit issue

GektorUA, 2019-10-14:
A few weeks on latest master branch with QCA9880 (with ath10k and ath10k-ct) i can't configure it from LUCI:

When i click on ADD or EDIT from Firefox, i get:
[image]

From Chrome or Windows Edge i get:
[image]

But when i enable it - open network is discovered and i can join, so - wireless card is working good, when i click on EDIT, i get:
[image]

LuCI Master (git-19.282.66222-6fb5fd3) / OpenWrt SNAPSHOT r11215-bba6646b5c

GektorUA, 2019-10-15:
I forgot to write — system is x86-64, config:

`config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/c98491b1-328b-41d8-a6a4-1f4a70f3dfa1/pcia6a4:00/a6a4:00:00.0'
	option htmode 'VHT80'
	option disabled '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'`

jow-, 2019-10-15:
Please provide the output of `ubus call iwinfo info '{ "device": "radio0" }'` and `ubus call iwinfo info '{ "device": "wlan0" }'`

GektorUA, 2019-10-15:
root@OpenWrt:~# ubus call iwinfo info '{ "device": "radio0" }'
Command failed: Not found

root@OpenWrt:~# ubus call iwinfo info '{ "device": "wlan0" }'
{
        "phy": "phy0",
        "ssid": "OpenWrt",
        "bssid": "04:F0:21:0F:4A:16",
        "country": "US",
        "mode": "Master",
        "channel": 44,
        "frequency": 5220,
        "frequency_offset": 0,
        "txpower": 23,
        "txpower_offset": 0,
        "quality_max": 70,
        "noise": -100,
        "encryption": {
                "enabled": true,
                "wpa": [
                        2
                ],
                "authentication": [
                        "psk"
                ],
                "ciphers": [
                        "ccmp"
                ]
        },
        "htmodes": [
                "HT20",
                "HT40",
                "VHT20",
                "VHT40",
                "VHT80"
        ],
        "hwmodes": [
                "ac",
                "b",
                "g",
                "n"
        ],
        "hardware": {
                "id": [
                        5772,
                        60,
                        0,
                        0
                ],
                "name": "Qualcomm Atheros QCA9880"
        }
}

jow-, 2019-10-15:
The first failing command is triggering the ui problem.

Can you please install `strace` and provide the output of `strace iwinfo radio0 freqlist` ?

jow-, 2019-10-15:
The underlying issue is likely fixed now by https://github.com/openwrt/openwrt/commit/57b834281b586839b5e2cb00d7907de50c68ebcc

GektorUA, 2019-10-15:
Thank you, now all works good! )

closed 2019-10-15 by GektorUA