openwrt/luci #3233: [Feature request] Minimal LUCI for (IOT devices for example)

BLLuis, 2019-10-23:
I know that OpenWRT is for routers, in fact one of my routers got version 15.05 installed and I'm waiting the end of the warranty of my last router to move to OpenWRT, but I have more than 5 devices with OpenWRT for my own use, IPCams, WirelessSPK, Nas and Home Automation, all of them dont act as routers, just a simple IOT device or as they are named IPCam.
Some of them have LUCI installed, some others not, why?
Because LUCI have so dependencies for the predefined options in menuconfig, I mean, iptables, firewall and some other network functions, almost for routers, and it's ok, but they aren't needed on some devices (None in my case, except on routers).
I know that we can strip LUCI and install a very few packages using some scripts or removing them by hand, but is a nightmare for 5 devices to update.
A useful configuration on the menuconfig->Luci entries should be add something like LUCI-ADMIN-MIN without iptables/firewall/pppoe.

Some kind of LUCI to configure a basic device and not the full functions of a router.

Thanks
Luis

hmh, 2020-01-26:
A very lightweight RPC (but with proper auth, maybe based in usign or some sort of HMAC) that would be easy to talk to from an iOS/Android APP seems to be in order, since luci-base + uhttpd would already be large enough to be unusable on 4MiB FLASH.

We've been trying to get by around here using SSH into uci, with a reboot trigger when uci commit succeeds.  But that's very unoptimal.

systemcrash, 2025-03-29:
What happened to the minimal one you worked on, @stokito ?