openwrt/luci #3237: luci-app-openvpn - git-19.296.26319-ca781af-1

gipsea, 2019-10-23:
Good morning.

I'm running openwrt 18.06 and yesterday I figured out there was an update of the openvpn package which I did.

Since then, anytime I try to edit one of the openvpn profile I'm redirect to a page with only this message:

> Failed to execute cbi dispatcher target for entry '/admin/services/openvpn/basic/custom_config'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/openvpn-basic.lua:37: attempt to index global 'fs' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/cbi/openvpn-basic.lua:37: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:841: in function </usr/lib/lua/luci/dispatcher.lua:835>

As I've been playing a lot with the configs I thought I mess up the system so at teh end I performed a total reset flashing with serial and tftp. As soon as I installed the package I received the same msg

At this point I think the problem is beyond me messing up ahahaha

If more info are required please let me know

cheers

mauro

GFTwrt, 2019-10-23:
Hello,
This is related with https://github.com/openwrt/luci/commit/ca781af2f8052d2469d3e6f533208c9e6e9648d4.

I requested https://github.com/openwrt/luci/issues/3163
But there were some additional stuff build around

feckert, 2019-10-24:
Thanks for the feedback. That was my mistake when cherry-pick didn't pay attention that **fs** wasn't imported with reqrire. With the other branches this was already changed.
Since I don't have a device with the firmware openwrt-18.06, can someone please change the following change manually in the source and check if that works?
Replace the following line in **/usr/lib/lua/luci/model/cbi/openvpn-basic.lua**
```
local has_ipv6 = fs.access("/proc/net/ipv6_route")
```
with
```
local has_ipv6 = nixio.fs.access("/proc/net/ipv6_route")
```
If so, I would check it in.

feckert, 2019-10-24:
Made a quick check by coping the file to a router with master branch.
And as discribed by @gipsea it does not work.
After the change it works again.
I did a push to openwrt-18.06 with the changes.
See https://github.com/openwrt/luci/commit/fbefeed9731ba4dc5c2269bc9ffc29d7541bb30d

@gipsea Thanks again for reporting good job