openwrt/luci #3243: vpnbypass action call violates dispatcher

ananas77777, 2019-10-28:
Mon Oct 28 19:01:21 2019 daemon.err uhttpd[909]: /usr/lib/lua/luci/dispatcher.lua:397: Access Violation
Mon Oct 28 19:01:21 2019 daemon.err uhttpd[909]: The page at 'admin/services/vpnbypass/action/enable/' has no parent node so the access to this location has been denied.
Mon Oct 28 19:01:21 2019 daemon.err uhttpd[909]: This is a software bug, please report this message at https://github.com/openwrt/luci/issues
Mon Oct 28 19:01:21 2019 daemon.err uhttpd[909]: stack traceback:
Mon Oct 28 19:01:21 2019 daemon.err uhttpd[909]: 	[C]: in function 'assert'
Mon Oct 28 19:01:21 2019 daemon.err uhttpd[909]: 	/usr/lib/lua/luci/dispatcher.lua:397: in function 'dispatch'
Mon Oct 28 19:01:21 2019 daemon.err uhttpd[909]: 	/usr/lib/lua/luci/dispatcher.lua:144: in function </usr/lib/lua/luci/dispatcher.lua:143>

feckert, 2019-10-28:
What version?

ananas77777, 2019-10-28:
That's latest from souce 1.3.1-4

Only sorted when I changed the location to force on admin/services tab not admin/vpn

feckert, 2019-10-28:
Did you reboot your system so that the ccache of LuCI is new?

ananas77777, 2019-10-28:
Yeah rebooted and the issue persists up until I changed the entry to be admin/services, This problem is that you call the admin/services entry even if admin/vpn is in place

feckert, 2019-10-28:
Could this be the root cause of your problem?
https://github.com/openwrt/luci/blob/7707b679e20df9806fbcbc58d12116f9ee9a3850/applications/luci-app-vpnbypass/luasrc/controller/vpnbypass.lua#L7-L10

If this is the problem we have to fix also the line https://github.com/openwrt/luci/blob/7707b679e20df9806fbcbc58d12116f9ee9a3850/applications/luci-app-vpnbypass/luasrc/view/vpnbypass/js.htm#L35 @stangri

ananas77777, 2019-10-28:
That's exactly is the issue, You need to build url for admin/vpn as well and define " entry({"admin", "services", "vpnbypass", "action"}, call("vpnbypass_action"), nil).leaf = true " or " entry({"admin", "vpn", "vpnbypass", "action"}, call("vpnbypass_action"), nil).leaf = true " based on the "if statement"

stangri, 2019-10-28:
@ananas77777 could you please try the updated luci-app-vpnbypass from [my repo](https://stangri.github.io/openwrt-repo/) before I send the PR?

I don't have a device I can test 19.07/snapshot on. Thanks!

ananas77777, 2019-10-28:
All good now, You can go ahead with the PR

closed 2019-10-29 by feckert

feckert, 2019-10-29:
@ananas77777 thanks for reporting this issue
@stangri thanks for the fast fix