openwrt/luci #3254: Feature request: DummySection

Ansuel, 2019-11-02:
@jow- 
I notice that in the new api is not present a way to create DummySection like we do with DummyValues. This can be usefull to create section that contains read only data for example version information, buttons or service status... 

Can this be added to form.js? 

For now this can be bypassed by setting a NamedSection that contains actually configs but it's not that clean.

ysc3839, 2019-11-04:
Why not using DummyValues? Can you provide any screenshot showing it's use cases?

Ansuel, 2019-11-04:
One example is this...

[image]

All the data present in this section are get from a separate rpcd request and they are not related to any uci section

To actually display DummyValues i'm forced to use a NamedSection although i don't use any uci config

ysc3839, 2019-11-04:
@Ansuel Consider using `JSONMap`.
https://github.com/openwrt/luci/blob/4e9f2d3f1ef21262e8c009579d235d355fbd467a/modules/luci-base/htdocs/luci-static/resources/form.js#L432-L440

Ansuel, 2019-11-04:
@ysc3839 mh i already have a form.Map defined

Can you give me a more specific example? 

Take notice that other section actually use uci config

ysc3839, 2019-11-04:
@Ansuel IIRC, you can merge two maps by:
```
return Promise.all([
  jsonmap.render(),
  map.render()
]);
```

jow-, 2019-11-04:
I'll see if I can look into it during the next few days. It probably makes sense to provide a dedicated `JSONSection` which can be used inside normal UCI based maps.

jow-, 2020-02-23:
I never got around to implement it but I'd say `JSONMap()` is the most flexible solution covering all cases.

closed 2020-02-23 by jow-