openwrt/luci #3259: Error  No related RPC reply

xseven007, 2019-11-04:
When I try to login in openwrt on boswer(like IE, and Chrome, Firefox and etc). It was show:
Error

No related RPC reply

I was used luci-nginx.. And I found the error-log of nginx in the /tmp/log/nginx, It is 
2019/11/04 19:45:45 [error] 3691#0: *23 Request generated error: Access denied, client: 192.168.1.100, server: localhost, request: "POST /ubus/?1572867944951 HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/cgi-bin/luci/"
2019/11/04 19:45:48 [error] 3691#0: *32 Request generated error: Access denied, client: 192.168.1.100, server: localhost, request: "POST /ubus/?1572867947293 HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/cgi-bin/luci/"
2019/11/04 19:45:48 [error] 3691#0: *33 Request generated error: Access denied, client: 192.168.1.100, server: localhost, request: "POST /ubus/?1572867947309 HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/cgi-bin/luci/"
2019/11/04 19:45:48 [error] 3691#0: *38 Request generated error: Access denied, client: 192.168.1.100, server: localhost, request: "POST /ubus/?1572867947354 HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/cgi-bin/luci/"
2019/11/04 19:45:48 [error] 3691#0: *39 Request generated error: Access denied, client: 192.168.1.100, server: localhost, request: "POST /ubus/?1572867947369 HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/cgi-bin/luci/"

jow-, 2019-11-04:
Close and reopen your browser tab or clear your local session storage.

xseven007, 2019-11-04:
@jow- I have done. But nothing change. I try to login with another Computer or Phone. It still show this message.

panda-mute, 2019-11-04:
@jow- it is the same issue in my test, it may nginx need some adaption？

jow-, 2019-11-04:
No idea about nginx, sorry

panda-mute, 2019-11-04:
@Ansuel Can you take a look?

Ansuel, 2019-11-04:
Can you give me Top output
Also try restart rpcd

xseven007, 2019-11-04:
@Ansuel I have tried to restart rpcd, but it doesn't work, and it still show this message: 
"Error
No related RPC reply"
on the top of page.

Ansuel, 2019-11-04:
Can you give me the request and the response from the browser? (Open Dev
console, go to network and give me the request and response of any /ubus/ )

xseven007, 2019-11-04:
@Ansuel Okey. I found this from Chrome.
Request URL: http://192.168.1.1/ubus/?1572877212310
Request Method: POST
Status Code: 200 OK
Remote Address: 192.168.1.1:80
Referrer Policy: no-referrer-when-downgrade

Response is:
{"jsonrpc":"2.0","id":null,"error":{"code":-32002,"message":"Access denied"}}

And the error-log with nginx is 
"Request generated error: Access denied, client: 192.168.1.100, server: localhost, request: "POST /ubus/?1572877212310 HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/cgi-bin/luci/""

Ansuel, 2019-11-04:
Right click on that.... Copy link an select curl bash and post here

xseven007, 2019-11-04:
curl 'http://192.168.1.1/ubus/?1572909239114' -H 'Connection: keep-alive' -H 'Pragma: no-cache' -H 'Cache-Control: no-cache' -H 'Origin: http://192.168.1.1' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/78.0.3904.70 Safari/537.36' -H 'DNT: 1' -H 'Content-Type: application/json' -H 'Accept: */*' -H 'Referer: http://192.168.1.1/cgi-bin/luci/' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: zh-CN,zh;q=0.9,en;q=0.8,en-US;q=0.7' --data-binary '[{"jsonrpc":"2.0","id":1,"method":"call","params":["00000000000000000000000000000000","luci","getFeatures",{}]}]' --compressed --insecure

Ansuel, 2019-11-04:
on my side it does work as it should... may i know your system and nginx version from opkg?

also your image is built by yourself or from openwrt?

xseven007, 2019-11-04:
nginx	1.16.1-4	
nginx-mod-luci	1.16.1-4	
luci-nginx	git-19.308.21829-2ff7d24-1

But the build config for the openwrt is too long to commit, because there is only 65535 that can commit

panda-mute, 2019-11-05:
> on my side it does work as it should... may i know your system and nginx version from opkg?
> 
> also your image is built by yourself or from openwrt?

This issue is caused by jow's modification of luci recently, there are many changes of luci. If you build openwrt with all latest version, then you can reproduce this issue.

Ansuel, 2019-11-05:
Ok I will update my image and I will test

Ansuel, 2019-11-05:
tested with latest commits and it does work well... 
We wait for other issue like this but for now i can't replicate this...

leonardpitzu, 2019-11-05:
same for me...
luci.js:100 - this is where it points to

EXSERENS, 2019-11-07:
in my test this is related to your webserver's permisson. running nginx as nobody:nogroup will reproduce this problem. in before, use nobody:nogroup is okay but now dont.
I think there's no special reason to run nginx as nobody, it still can operate uwsgi and that uwsgi must run as root. so that maynot improve your security.

Ansuel, 2019-11-08:
Didn't think about this... I don't think there is a way to solve this
Am I wrong?

EXSERENS, 2019-11-08:
> […](#)

if uhttpd also run as root, it maybe no need to fix.
or it can be solve by add permisson? like allow www-data operate uwsgi. (i think it related to uwsgi, but not sure)

Ansuel, 2019-11-08:
No with ubus module nginx communicate directly with ubus and as it doesn't
have permissions everything fails

crackself, 2019-11-10:
I got the same error, update to branch master or openwnrt-19.07. Tested on raspi3 B+, by using nginx as web server.

Ansuel, 2019-11-11:
Can someone test with latest commit? There was a problem with list method

panda-mute, 2019-11-11:
No luck, I still met the same issue with your nginx  ubus module fix.

leonardpitzu, 2019-11-11:
> Can someone test with latest commit? There was a problem with list method

tested now - seems to work ok for me. @panda-mute do you have any changes that might affect the server? i've got a clean build and it works ok for me. maybe we shall wait for others to test out.

Ykidia, 2019-11-12:
I am testing it right now too. Not so good. My settings is clean (sysupgraded with -n option).
[image]
[image]

Ansuel, 2019-11-12:
@Ykidia can you post here your nginx conf ? 
also the module installed on luci ?

Ykidia, 2019-11-12:
/etc/nginx/nginx.conf : 
`
user  root;
worker_processes  4;

#error_log  logs/error.log;
#error_log  logs/error.log  notice;
#error_log  logs/error.log  info;

pid        /var/run/nginx.pid;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    default_type  application/octet-stream;

    sendfile on;
    keepalive_timeout 0;

    client_body_buffer_size 10K;
    client_header_buffer_size 1k;
    client_max_body_size 1G;
    large_client_header_buffers 2 1k;

    gzip on;
    gzip_http_version 1.1;
    gzip_vary on;
    gzip_comp_level 1;
    gzip_proxied any;
    gzip_types text/plain text/css application/json application/x-javascript text/xml application/xml application/xml+rss text/javascript application/javascript image/svg+xml;

    root /www;

    server {
        listen 80 default_server;
        listen [::]:80 default_server;
        server_name _;
        return 301 https://$host$request_uri;
    }

    server {
        listen 443 ssl default_server;
        listen [::]:443 ssl default_server;
        server_name  localhost;

        ssl_protocols TLSv1 TLSv1.1 TLSv1.2;
        ssl_prefer_server_ciphers on;
        ssl_ciphers "EECDH+ECDSA+AESGCM:EECDH+aRSA+AESGCM:EECDH+ECDSA+SHA384:EECDH+ECDSA+SHA256:EECDH+aRSA+SHA384:EECDH+aRSA+SHA256:EECDH:DHE+AESGCM:DHE:!RSA!aNULL:!eNULL:!LOW:!RC4:!3DES:!MD5:!EXP:!PSK:!SRP:!DSS:!CAMELLIA:!SEED";
        ssl_session_tickets off;

        ssl_certificate /etc/nginx/nginx.cer;
        ssl_certificate_key /etc/nginx/nginx.key;

        location ~* .(jpg|jpeg|png|gif|ico|css|js)$ {
            expires 365d;
        }

        include luci_uwsgi.conf;

    }

    include /etc/nginx/conf.d/*.conf;
}
`

Which module?

Ansuel, 2019-11-12:
`opkg list luci*`

Ykidia, 2019-11-12:
root@OpenWrt:~# opkg list luci*
luci-app-adblock - git-19.315.33243-0731f7e-1
luci-app-aria2 - git-19.315.33243-0731f7e-1
luci-app-ddns - 2.4.9-7
luci-app-dnscrypt-proxy - git-19.315.33243-0731f7e-1
luci-app-firewall - git-19.315.33243-0731f7e-1
luci-app-hd-idle - git-19.315.33243-0731f7e-1
luci-app-https_dns_proxy - git-19.315.33243-0731f7e-5
luci-app-minidlna - git-19.315.33243-0731f7e-1
luci-app-mjpg-streamer - git-19.315.33243-0731f7e-1
luci-app-mwan3 - git-19.315.33243-0731f7e-1
luci-app-nut - git-19.315.33243-0731f7e-1
luci-app-openvpn - git-19.315.33243-0731f7e-1
luci-app-opkg - git-19.315.33243-0731f7e-1
luci-app-p910nd - git-19.315.33243-0731f7e-1
luci-app-samba4 - git-19.315.33243-0731f7e-1
luci-app-sqm - 1.4.0-2
luci-app-transmission - git-19.315.33243-0731f7e-1
luci-app-udpxy - git-19.315.33243-0731f7e-1
luci-app-upnp - git-19.315.33243-0731f7e-1
luci-app-vlmcsd - 1.0.2-1
luci-app-watchcat - git-19.315.33243-0731f7e-1
luci-app-wireguard - git-19.315.33243-0731f7e-1
luci-app-wol - git-19.315.33243-0731f7e-1
luci-base - git-19.315.33243-0731f7e-1
luci-compat - git-19.315.33243-0731f7e-1
luci-i18n-adblock-ru - git-19.315.33243-0731f7e-1
luci-i18n-adblock-uk - git-19.315.33243-0731f7e-1
luci-i18n-aria2-ru - git-19.315.33243-0731f7e-1
luci-i18n-aria2-uk - git-19.315.33243-0731f7e-1
luci-i18n-base-ru - git-19.315.33243-0731f7e-1
luci-i18n-base-uk - git-19.315.33243-0731f7e-1
luci-i18n-ddns-ru - 2.4.9-7
luci-i18n-ddns-uk - 2.4.9-7
luci-i18n-dnscrypt-proxy-ru - git-19.315.33243-0731f7e-1
luci-i18n-dnscrypt-proxy-uk - git-19.315.33243-0731f7e-1
luci-i18n-firewall-ru - git-19.315.33243-0731f7e-1
luci-i18n-firewall-uk - git-19.315.33243-0731f7e-1
luci-i18n-hd-idle-ru - git-19.315.33243-0731f7e-1
luci-i18n-hd-idle-uk - git-19.315.33243-0731f7e-1
luci-i18n-https_dns_proxy-ru - git-19.315.33243-0731f7e-5
luci-i18n-https_dns_proxy-uk - git-19.315.33243-0731f7e-5
luci-i18n-minidlna-ru - git-19.315.33243-0731f7e-1
luci-i18n-minidlna-uk - git-19.315.33243-0731f7e-1
luci-i18n-mjpg-streamer-ru - git-19.315.33243-0731f7e-1
luci-i18n-mjpg-streamer-uk - git-19.315.33243-0731f7e-1
luci-i18n-mwan3-ru - git-19.315.33243-0731f7e-1
luci-i18n-mwan3-uk - git-19.315.33243-0731f7e-1
luci-i18n-nut-ru - git-19.315.33243-0731f7e-1
luci-i18n-nut-uk - git-19.315.33243-0731f7e-1
luci-i18n-openvpn-ru - git-19.315.33243-0731f7e-1
luci-i18n-openvpn-uk - git-19.315.33243-0731f7e-1
luci-i18n-opkg-ru - git-19.315.33243-0731f7e-1
luci-i18n-opkg-uk - git-19.315.33243-0731f7e-1
luci-i18n-p910nd-ru - git-19.315.33243-0731f7e-1
luci-i18n-p910nd-uk - git-19.315.33243-0731f7e-1
luci-i18n-samba4-ru - git-19.315.33243-0731f7e-1
luci-i18n-samba4-uk - git-19.315.33243-0731f7e-1
luci-i18n-transmission-ru - git-19.315.33243-0731f7e-1
luci-i18n-transmission-uk - git-19.315.33243-0731f7e-1
luci-i18n-udpxy-ru - git-19.315.33243-0731f7e-1
luci-i18n-udpxy-uk - git-19.315.33243-0731f7e-1
luci-i18n-upnp-ru - git-19.315.33243-0731f7e-1
luci-i18n-upnp-uk - git-19.315.33243-0731f7e-1
luci-i18n-watchcat-ru - git-19.315.33243-0731f7e-1
luci-i18n-watchcat-uk - git-19.315.33243-0731f7e-1
luci-i18n-wireguard-ru - git-19.315.33243-0731f7e-1
luci-i18n-wireguard-uk - git-19.315.33243-0731f7e-1
luci-i18n-wol-ru - git-19.315.33243-0731f7e-1
luci-i18n-wol-uk - git-19.315.33243-0731f7e-1
luci-lib-httpprotoutils - git-19.315.33243-0731f7e-1
luci-lib-ip - git-19.315.33243-0731f7e-1
luci-lib-ipkg - git-19.315.33243-0731f7e-1
luci-lib-jsonc - git-19.315.33243-0731f7e-1
luci-lib-nixio - git-19.315.33243-0731f7e-1
luci-mod-admin-full - git-19.315.33243-0731f7e-1
luci-mod-network - git-19.315.33243-0731f7e-1
luci-mod-status - git-19.315.33243-0731f7e-1
luci-mod-system - git-19.315.33243-0731f7e-1
luci-proto-3g - git-19.315.33243-0731f7e-1
luci-proto-ipv6 - git-19.315.33243-0731f7e-1
luci-proto-ppp - git-19.315.33243-0731f7e-1
luci-proto-relay - git-19.315.33243-0731f7e-1
luci-proto-wireguard - git-19.315.33243-0731f7e-1
luci-ssl-nginx - git-19.315.33243-0731f7e-1
luci-theme-bootstrap - git-19.315.33243-0731f7e-1
luci-theme-material - git-19.315.33243-0731f7e-1
luci-theme-openwrt - git-19.315.33243-0731f7e-1

Ykidia, 2019-11-12:
Yes, there are a lot of modules, but there were no problems before (at least with default settings)...

Ansuel, 2019-11-12:
can you test me if network -> interface works ?

Ykidia, 2019-11-12:
Yes, this page is works, but...
[image]

Ykidia, 2019-11-12:
I can post full build (sysupgrade binary) if necessary.

Ansuel, 2019-11-12:
will update to latest commit and retest 
just can you give me the request url ( inspect element, tab network, right click on the request with ?41937192 (random numbers) and select curl bash )

Ykidia, 2019-11-12:
System => System : 
```
curl 'https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'Connection: keep-alive' -H 'Upgrade-Insecure-Requests: 1' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' -H 'Sec-Fetch-Mode: navigate' -H 'Sec-Fetch-User: ?1' -H 'Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3' -H 'Sec-Fetch-Site: same-origin' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/' -H 'Accept-Encoding: gzip, deflate, br' -H 'Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7' -H 'Cookie: sysauth=23b888e8788f88c9e9d28b5358cd665d' --compressed --insecure ;
curl 'https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/translations?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: no-cors' -H 'Sec-Fetch-Site: same-origin' -H 'Accept-Encoding: gzip, deflate, br' -H 'Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' -H 'Accept: */*' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'Cookie: sysauth=23b888e8788f88c9e9d28b5358cd665d' -H 'Connection: keep-alive' --compressed --insecure ;
curl 'https://ykidia.no-ip.info:3443/luci-static/resources/cbi.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/resources/xhr.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/material/brand.png' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/resources/promis.min.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/resources/luci.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/material/js/jquery.min.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/material/js/script.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/material/custom.css' -H 'Sec-Fetch-Mode: no-cors' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' --compressed ;
curl 'https://ykidia.no-ip.info:3443/luci-static/resources/ui.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: cors' -H 'Sec-Fetch-Site: same-origin' -H 'Accept-Encoding: gzip, deflate, br' -H 'Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' -H 'Accept: */*' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'Connection: keep-alive' --compressed --insecure ;
curl 'https://ykidia.no-ip.info:3443/luci-static/resources/rpc.js?v=git-19.315.33243-0731f7e' -H 'Sec-Fetch-Mode: cors' -H 'Sec-Fetch-Site: same-origin' -H 'Accept-Encoding: gzip, deflate, br' -H 'Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' -H 'Accept: */*' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'Connection: keep-alive' --compressed --insecure ;
[... cut ...] ication/json' -H 'Accept: */*' -H 'Referer: https://ykidia.no-ip.info:3443/cgi-bin/luci/admin/system/system' -H 'Connection: keep-alive' --data-binary '[{"jsonrpc":"2.0","id":7,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","uci","changes",{}]},{"jsonrpc":"2.0","id":8,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","uci","changes",{}]},{"jsonrpc":"2.0","id":9,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","uci","changes",{}]}]' --compressed --insecure
```

Ykidia, 2019-11-12:
Network => Interfaces : 
`curl 'https://ykidia.MYDOMAIN:MYPORT/ubus/?1573521962542' -H 'Sec-Fetch-Mode: cors' -H 'Sec-Fetch-Site: same-origin' -H 'Origin: https://ykidia.no-ip.info:3443' -H 'Accept-Encoding: gzip, deflate, br' -H 'Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36 OPR/64.0.3417.92' -H 'Content-Type: application/json' -H 'Accept: */*' -H 'Referer: https://ykidia.MYDOMAIN:MYPORT/cgi-bin/luci/admin/network/network' -H 'Connection: keep-alive' --data-binary '[{"jsonrpc":"2.0","id":1,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","luci","getFeatures",{}]},{"jsonrpc":"2.0","id":2,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","network.interface","dump",{}]},{"jsonrpc":"2.0","id":3,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","luci-rpc","getBoardJSON",{}]},{"jsonrpc":"2.0","id":4,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","luci-rpc","getNetworkDevices",{}]},{"jsonrpc":"2.0","id":5,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","luci-rpc","getWirelessDevices",{}]},{"jsonrpc":"2.0","id":6,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","luci-rpc","getHostHints",{}]},{"jsonrpc":"2.0","id":7,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","network","get_proto_handlers",{}]},{"jsonrpc":"2.0","id":8,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","uci","get",{"config":"network"}]},{"jsonrpc":"2.0","id":9,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","uci","get",{"config":"wireless"}]},{"jsonrpc":"2.0","id":10,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","uci","get",{"config":"luci"}]},{"jsonrpc":"2.0","id":11,"method":"call","params":["23b888e8788f88c9e9d28b5358cd665d","uci","changes",{}]}]' --compressed --insecure`

Ykidia, 2019-11-12:
These two pages are from my home router with some non-default config (but nginx config is default, however)

crackself, 2019-11-12:
Hey, the new update fix this PRC error, on raspi3 B+
Test on LuCI openwrt-19.07 branch (git-19.315.33349-ae61ecb) / OpenWrt 19.07-SNAPSHOT r10194+485-c53f62b111

Ykidia, 2019-11-12:
> the new update
> OpenWrt 19.07-SNAPSHOT r10194+485-c53f62b111
That (**r10194**) is NOT a "new update".

> OpenWrt SNAPSHOT, r11486-6becc37f33
And this (**r11486**) IS a REALLY NEW update:
```
root@OpenWrt:~# opkg list luci*
luci-app-adblock - git-19.316.39109-95bfb3f-1
luci-app-aria2 - git-19.316.39109-95bfb3f-1
luci-app-ddns - git-19.316.39109-95bfb3f-1
luci-app-dnscrypt-proxy - git-19.316.39109-95bfb3f-1
luci-app-firewall - git-19.316.39109-95bfb3f-1
luci-app-hd-idle - git-19.316.39109-95bfb3f-1
luci-app-https_dns_proxy - git-19.316.39109-95bfb3f-5
luci-app-minidlna - git-19.316.39109-95bfb3f-1
luci-app-mjpg-streamer - git-19.316.39109-95bfb3f-1
luci-app-mwan3 - git-19.316.39109-95bfb3f-1
luci-app-nut - git-19.316.39109-95bfb3f-1
luci-app-openvpn - git-19.316.39109-95bfb3f-1
luci-app-opkg - git-19.316.39109-95bfb3f-1
luci-app-p910nd - git-19.316.39109-95bfb3f-1
luci-app-samba4 - git-19.316.39109-95bfb3f-1
luci-app-sqm - 1.4.0-2
luci-app-transmission - git-19.316.39109-95bfb3f-1
luci-app-udpxy - git-19.316.39109-95bfb3f-1
luci-app-upnp - git-19.316.39109-95bfb3f-1
luci-app-vlmcsd - 1.0.2-1
luci-app-watchcat - git-19.316.39109-95bfb3f-1
luci-app-wireguard - git-19.316.39109-95bfb3f-1
luci-app-wol - git-19.316.39109-95bfb3f-1
luci-base - git-19.316.39109-95bfb3f-1
luci-compat - git-19.316.39109-95bfb3f-1
luci-i18n-adblock-ru - git-19.316.39109-95bfb3f-1
luci-i18n-adblock-uk - git-19.316.39109-95bfb3f-1
luci-i18n-aria2-ru - git-19.316.39109-95bfb3f-1
luci-i18n-aria2-uk - git-19.316.39109-95bfb3f-1
luci-i18n-base-ru - git-19.316.39109-95bfb3f-1
luci-i18n-base-uk - git-19.316.39109-95bfb3f-1
luci-i18n-ddns-ru - git-19.316.39109-95bfb3f-1
luci-i18n-ddns-uk - git-19.316.39109-95bfb3f-1
luci-i18n-dnscrypt-proxy-ru - git-19.316.39109-95bfb3f-1
luci-i18n-dnscrypt-proxy-uk - git-19.316.39109-95bfb3f-1
luci-i18n-firewall-ru - git-19.316.39109-95bfb3f-1
luci-i18n-firewall-uk - git-19.316.39109-95bfb3f-1
luci-i18n-hd-idle-ru - git-19.316.39109-95bfb3f-1
luci-i18n-hd-idle-uk - git-19.316.39109-95bfb3f-1
luci-i18n-https_dns_proxy-ru - git-19.316.39109-95bfb3f-5
luci-i18n-https_dns_proxy-uk - git-19.316.39109-95bfb3f-5
luci-i18n-minidlna-ru - git-19.316.39109-95bfb3f-1
luci-i18n-minidlna-uk - git-19.316.39109-95bfb3f-1
luci-i18n-mjpg-streamer-ru - git-19.316.39109-95bfb3f-1
luci-i18n-mjpg-streamer-uk - git-19.316.39109-95bfb3f-1
luci-i18n-mwan3-ru - git-19.316.39109-95bfb3f-1
luci-i18n-mwan3-uk - git-19.316.39109-95bfb3f-1
luci-i18n-nut-ru - git-19.316.39109-95bfb3f-1
luci-i18n-nut-uk - git-19.316.39109-95bfb3f-1
luci-i18n-openvpn-ru - git-19.316.39109-95bfb3f-1
luci-i18n-openvpn-uk - git-19.316.39109-95bfb3f-1
luci-i18n-opkg-ru - git-19.316.39109-95bfb3f-1
luci-i18n-opkg-uk - git-19.316.39109-95bfb3f-1
luci-i18n-p910nd-ru - git-19.316.39109-95bfb3f-1
luci-i18n-p910nd-uk - git-19.316.39109-95bfb3f-1
luci-i18n-samba4-ru - git-19.316.39109-95bfb3f-1
luci-i18n-samba4-uk - git-19.316.39109-95bfb3f-1
luci-i18n-transmission-ru - git-19.316.39109-95bfb3f-1
luci-i18n-transmission-uk - git-19.316.39109-95bfb3f-1
luci-i18n-udpxy-ru - git-19.316.39109-95bfb3f-1
luci-i18n-udpxy-uk - git-19.316.39109-95bfb3f-1
luci-i18n-upnp-ru - git-19.316.39109-95bfb3f-1
luci-i18n-upnp-uk - git-19.316.39109-95bfb3f-1
luci-i18n-watchcat-ru - git-19.316.39109-95bfb3f-1
luci-i18n-watchcat-uk - git-19.316.39109-95bfb3f-1
luci-i18n-wireguard-ru - git-19.316.39109-95bfb3f-1
luci-i18n-wireguard-uk - git-19.316.39109-95bfb3f-1
luci-i18n-wol-ru - git-19.316.39109-95bfb3f-1
luci-i18n-wol-uk - git-19.316.39109-95bfb3f-1
luci-lib-httpprotoutils - git-19.316.39109-95bfb3f-1
luci-lib-ip - git-19.316.39109-95bfb3f-1
luci-lib-ipkg - git-19.316.39109-95bfb3f-1
luci-lib-jsonc - git-19.316.39109-95bfb3f-1
luci-lib-nixio - git-19.316.39109-95bfb3f-1
luci-mod-admin-full - git-19.316.39109-95bfb3f-1
luci-mod-network - git-19.316.39109-95bfb3f-1
luci-mod-status - git-19.316.39109-95bfb3f-1
luci-mod-system - git-19.316.39109-95bfb3f-1
luci-proto-3g - git-19.316.39109-95bfb3f-1
luci-proto-ipv6 - git-19.316.39109-95bfb3f-1
luci-proto-ppp - git-19.316.39109-95bfb3f-1
luci-proto-relay - git-19.316.39109-95bfb3f-1
luci-proto-wireguard - git-19.316.39109-95bfb3f-1
luci-ssl-nginx - git-19.316.39109-95bfb3f-1
luci-theme-bootstrap - git-19.316.39109-95bfb3f-1
luci-theme-material - git-19.316.39109-95bfb3f-1
luci-theme-openwrt - git-19.316.39109-95bfb3f-1

```
Another browser, same errors, even more (page Status => Overview, infinite "Loading view" progress):
```
RPCError
RPC call to luci.ddns/get_services_status failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-19.316.39109-95bfb3f:12:3)
```
```
RPCError
RPC call to luci/getFeatures failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-19.316.39109-95bfb3f:12:3)
```

SirToffski, 2019-11-12:
In my case this was resolved by removing the following from `/etc/nginx/luci_uwsgi.conf`

```
location /ubus {
        ubus_interpreter;
        ubus_socket_path /var/run/ubus.sock;
        ubus_parallel_req 2;
}
```

So now the `/etc/nginx/luci_uwsgi.conf` looks like:
```
# cat /etc/nginx/luci_uwsgi.conf.bak1
location /cgi-bin/luci {
		index  index.html;
		include uwsgi_params;
		uwsgi_param SERVER_ADDR $server_addr;
		uwsgi_modifier1 9;
		uwsgi_pass unix:////var/run/luci-webui.socket;
}
location ~ /cgi-bin/cgi-(backup|download|upload) {
		include uwsgi_params;
		uwsgi_param SERVER_ADDR $server_addr;
		uwsgi_modifier1 9;
		uwsgi_pass unix:////var/run/luci-cgi_io.socket;
}

location /luci-static {
}
```

Ansuel, 2019-11-12:
it's not a solution... just a workaround...

SirToffski, 2019-11-12:
I didn't say it was. The intention was to provide some information in case it helped determine the root case.

Ykidia, 2019-11-12:
This workaround looks pretty :-D
But... some settings gone and, after some time exploring LuCI pages I've got this: "Error
XHR request timed out". Hahaha...

Ansuel, 2019-11-12:
Then there is something wrong with your system... 
by removing /ubus in ningx you are using the normal way of calling rpc data

EXSERENS, 2019-11-13:
in latest Master build
```
RPC call to luci.ddns/get_services_status failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-19.316.39109-95bfb3f:12:3)
```

Ykidia, 2019-11-13:
Then I cleared browser cache (on the machine where I saw "Error
XHR request timed out"). And then started browsing clean router with default settings For the first time there was no any error. I could normally view Status => Overview, System => System, Network => Firewall, and so on. But there were some little limitations. For example, on System => System => General Settings there only one Timezone in list - UTC; on Network => Firewall => General Settings there were no software/hardware offloading settings.
But after 5 minutes I went to Status => Overview and then got "Error
XHR request timed out".
While writing this, I went again on some pages and returned to Status => Overview. No errors.
Then this error (XHR request timed out) is floating, perhaps it occurs when there are not enough resources (CPU). Then, for the present, we can ignore this error. And then it turns out that almost everything is fine with the workaround (but some few settings and elements, as I mentioned above).

But what happens when we turn on /ubus?

predators46, 2019-11-13:
Luci-theme-material Eror openwrt 19.07.0-rc1
[image]
[image]

zuzia-dev, 2019-11-13:
Test on  LuCI Master (git-19.316.39109-95bfb3f), Firefox 70.0.1 (with disabled DOM Storage). Linux Debian, XFCE. I cleared browser cache. When started browsing on clean router (I used the sysupgrade  with -n option) the problem is still here.

Status => Overview
[image]

Network => Interfaces, Hosts, Routes (also System, Autostart, Firmaware Upgrade).
[image]

Network => Switch
[image]

Ykidia, 2019-11-20:
What's wrong with LuCI for now? Even fully clean OpenWrt installation with uhttpd often gives an Error "XHR request timed out". I tried firmwares for UPVEL UR-313N4G (same as D-Link DIR-320 B1), UPVEL UR-337N4G (nearly the same as D-Link DIR-620 D1) and Xiaomi MiR3G. And soon I'l try to build new firmware for ASUS RT-AC68U (now it has fw built on 04 Nov 2019, with nginx instead of uhttpd, and there are no errors in LuCI).

Josua-SR, 2019-12-26:
> in my test this is related to your webserver's permisson. running nginx as nobody:nogroup will reproduce this problem. in before, use nobody:nogroup is okay but now dont.
> I think there's no special [...]

Just fyi I still have permissions problem nginx -> ubus with master branch as of today.
Changing nginx.conf to make it run as root works around the issue.

Josua-SR, 2019-12-26:
I did use strace on nginx: `strace -f /usr/sbin/nginx -c /etc/nginx/nginx.conf -g 'daemon off;'` to check if nginx is able to connect to the rpcd socket at /var/run/ubus.sock, and according to my log it is.

Perhaps rpcd is checking permissions that it did not check before?
Here an extract of the log:
```
[pid  8556] socket(AF_UNIX, SOCK_STREAM, 0) = 15
[pid  8556] fcntl64(15, F_GETFD)        = 0
[pid  8556] fcntl64(15, F_SETFD, FD_CLOEXEC) = 0
[pid  8556] connect(15, {sa_family=AF_UNIX, sun_path="/var/run/ubus.sock"}, 110) = 0
[pid  8556] read(15, "\0\0\0\0\233\374\232\213\0\0\0\4", 12) = 12
[pid  8556] read(15, "", 0)             = 0
[pid  8556] fcntl64(15, F_GETFL)        = 0x2 (flags O_RDWR)
[pid  8556] fcntl64(15, F_SETFL, O_RDWR|O_NONBLOCK|O_CLOEXEC) = 0
[pid  8559] sendmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\4\0\1\0\0\0\0", iov_len=8}, {iov_base="\0\0\0\f\2\0\0\10uci\0", iov_len=12}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 20
[pid  8559] poll([{fd=15, events=POLLIN|POLLERR}], 1, -1) = 1 ([{fd=15, revents=POLLIN}])
[pid  8559] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\1\0\1\0\0\0\0\0\0\0\f", iov_len=12}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 12
[pid  8559] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\10\0\0\0\0", iov_len=8}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 8
[pid  8559] sendmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\4\0\2\0\0\0\0", iov_len=8}, {iov_base="\0\0\0\20\2\0\0\fsession\0", iov_len=16}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 24
[pid  8559] poll([{fd=15, events=POLLIN|POLLERR}], 1, -1) = 1 ([{fd=15, revents=POLLIN}])
[pid  8559] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\1\0\2\0\0\0\0\0\0\0\f", iov_len=12}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 12
[pid  8559] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\10\0\0\0\0", iov_len=8}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 8
[pid  8559] sendmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\5\0\3\266\365\303\370", iov_len=8}, {iov_base="\0\0\0\204\3\0\0\10\266\365\303\370\4\0\0\vaccess\0\0\7\0\0l\203\0\09"..., iov_len=132}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 140
[pid  8559] poll([{fd=15, events=POLLIN|POLLERR}], 1, 30000) = 1 ([{fd=15, revents=POLLIN}])
[pid  8559] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\1\0\3\266\365\303\370\0\0\0\f", iov_len=12}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 12
[pid  8559] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\10\0\0\0\4", iov_len=8}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 8
[pid  8560] sendmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\4\0\4\0\0\0\0", iov_len=8}, {iov_base="\0\0\0\20\2\0\0\tfile\0\0\0\0", iov_len=16}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 24
[pid  8560] poll([{fd=15, events=POLLIN|POLLERR}], 1, -1) = 1 ([{fd=15, revents=POLLIN}])
[pid  8560] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\1\0\4\0\0\0\0\0\0\0\f", iov_len=12}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 12
[pid  8560] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\10\0\0\0\0", iov_len=8}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 8
[pid  8560] sendmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\4\0\5\0\0\0\0", iov_len=8}, {iov_base="\0\0\0\20\2\0\0\fsession\0", iov_len=16}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 24
[pid  8560] poll([{fd=15, events=POLLIN|POLLERR}], 1, -1) = 1 ([{fd=15, revents=POLLIN}])
[pid  8560] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\1\0\5\0\0\0\0\0\0\0\f", iov_len=12}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 12
[pid  8560] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\10\0\0\0\0", iov_len=8}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 8
[pid  8560] sendmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\5\0\6\0\0\0\0", iov_len=8}, {iov_base="\0\0\0\210\3\0\0\10\0\0\0\0\4\0\0\vaccess\0\0\7\0\0p\203\0\09"..., iov_len=136}], msg_iovlen=2,msg_controllen=0, msg_flags=0}, 0) = 144
[pid  8560] poll([{fd=15, events=POLLIN|POLLERR}], 1, 30000) = 1 ([{fd=15, revents=POLLIN}])
[pid  8560] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\1\0\6\0\0\0\0\0\0\0\f", iov_len=12}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 12
[pid  8560] recvmsg(15, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\10\0\0\0\4", iov_len=8}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 8
[pid  8556] close(15)                   = 0
```

And an extract from when rpcd was last updated before nivember 4th when this issue was opened:
```
commit c2675bb0cef373ff59fcc2dbd77471d244bbc774
Author: Jo-Philipp Wich <jo@mein.io>
Date:   Fri Nov 1 08:32:52 2019 +0100

    rpcd: update to latest Git HEAD
    
    d442d62 plugin: fix double free in finish callback
    ee26d83 main: exec_self: make clang analyzer happy
    90e40bd file: exec: properly free memory on error
    9ecfada uci: free configs list memory on return
    32fba36 exec: always call finish_cb to allow plugin to free up memory
    ca3e2d5 plugin: do not free method name separately
    02c6e1d exec: properly free memory on rpc_exec() error
    cc50263 plugin: exec: properly free memory on parse error
    bd0ed25 uci: reset uci_ptr flags when merging set operations
    37aa919 plugin: fix leaking invoked method name for exec plugins
    
    Signed-off-by: Jo-Philipp Wich <jo@mein.io>
```
Doesn't look suspicious though ...

avengermsoft, 2020-01-18:
Just fyi I still have permissions problem nginx -> ubus with master branch as of today.
Changing nginx.conf to make it run as root and removing /ubus in luci_uwsgi.conf  works around the issue.

avengermsoft, 2020-01-18:
[image]

locojohn, 2020-01-28:
For me it's not even nginx-related at all.  I am now ALWAYS having "XHR request timed out" in the status section of Luci web interface.  FYI: I am running uhttpd / luci with the latest master snapshot:

NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r12138-1e3bfbafd3"
OPENWRT_BOARD="ipq806x/generic"
OPENWRT_ARCH="arm_cortex-a15_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r12138-1e3bfbafd3"

jow-, 2020-01-28:
Using `luci-ssl`? What is reported by `opkg list_installed | grep luci` ?

locojohn, 2020-01-28:
> Using `luci-ssl`? What is reported by `opkg list_installed | grep luci` ?

Yes sir.  

OpenWrt-r12138-1e3bfbafd3 ~# opkg list_installed | grep luci
liblucihttp-lua - 2019-07-05-a34a17d5-1
liblucihttp0 - 2019-07-05-a34a17d5-1
luci - git-20.028.36882-b60b479-1
luci-app-adblock - git-20.028.36882-b60b479-1
luci-app-commands - git-20.028.36882-b60b479-1
luci-app-ddns - git-20.028.36882-b60b479-1
luci-app-diag-core - git-20.028.36882-b60b479-1
luci-app-firewall - git-20.028.36882-b60b479-1
luci-app-nlbwmon - git-20.028.36882-b60b479-1
luci-app-opkg - git-20.028.36882-b60b479-1
luci-app-samba - git-20.028.36882-b60b479-1
luci-app-sqm - 1.4.0-3
luci-app-statistics - git-20.028.36882-b60b479-1
luci-app-upnp - git-20.028.36882-b60b479-1
luci-app-wifischedule - git-20.028.36882-b60b479-1
luci-app-wol - git-20.028.36882-b60b479-1
luci-base - git-20.028.36882-b60b479-1
luci-compat - git-20.028.36882-b60b479-1
luci-lib-ip - git-20.028.36882-b60b479-1
luci-lib-ipkg - git-20.028.36882-b60b479-1
luci-lib-iptparser - git-20.028.36882-b60b479-1
luci-lib-jsonc - git-20.028.36882-b60b479-1
luci-lib-nixio - git-20.028.36882-b60b479-1
luci-mod-admin-full - git-20.028.36882-b60b479-1
luci-mod-network - git-20.028.36882-b60b479-1
luci-mod-status - git-20.028.36882-b60b479-1
luci-mod-system - git-20.028.36882-b60b479-1
luci-proto-ipv6 - git-20.028.36882-b60b479-1
luci-proto-ppp - git-20.028.36882-b60b479-1
luci-ssl-openssl - git-20.028.36882-b60b479-1
luci-theme-bootstrap - git-20.028.36882-b60b479-1
rpcd-mod-luci - 20191114

jow-, 2020-01-28:
Try without SSL, that is the likely culprit of your problems.

locojohn, 2020-01-28:
> Try without SSL, that is the likely culprit of your problems.

What kind of advice is that?   In 2020 without SSL?

jow-, 2020-01-28:
> What kind of advice is that?

An advice that aims to identify the root cause of your timeout problems. Depending on the outcome we can decide whether the SSL support is faulty and needs patching, whether changing `luci-ssl-openssl` to `luci-ssl` would help or whether the uhttpd configuration is not optimal.

[25 more replies]

closed 2020-02-23 by jow-