openwrt/luci #3269: 19.07rc1 6in4 interface is displayed twice on Status->Overview screen

hanipouspilot, 2019-11-06:
https://img15.lostpic.net/2019/11/06/0a160963418d8d1fa4d99e9605171136.png

hanipouspilot, 2019-11-06:
And also SQM doesn't open

```Failed to execute cbi dispatcher target for entry '/admin/network/sqm'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:938: module 'luci.cbi' not found:
	no field package.preload['luci.cbi']
	no file './luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi/init.lua'
	no file '/usr/lib/lua/luci/cbi.lua'
	no file '/usr/lib/lua/luci/cbi/init.lua'
	no file './luci/cbi.so'
	no file '/usr/lib/lua/luci/cbi.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:938: in function </usr/lib/lua/luci/dispatcher.lua:937>```

jow-, 2019-11-06:
You need to install `luci-compat` for `luci-app-sqm` to work.

hanipouspilot, 2019-11-06:
@jow-  The compat fixed SQM thx.

closed 2019-11-06 by jow-

yanyudalao, 2020-10-22:
> You need to install `luci-compat` for `luci-app-sqm` to work.

Unknown package 'luci-compat'.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-compat.

commit a8a76978 (2019-11-06): luci-base: network.js: prevent duplicate wan interface reporting

Fixes: #3269
commit c2be304f (2019-11-06): luci-base: network.js: prevent duplicate wan interface reporting

Fixes: #3269

(cherry picked from commit a8a76978298ba1cfddadbb48da73fa2abde82692)