openwrt/luci #3270: Openwrt-19.07 branch sysupgrade doesn't work

hanipouspilot, 2019-11-06:
After selecting a file and pressing upload nothing happens.
Interface is stuck on uploading 100% and real sysupgrade never starts.

jow-, 2019-11-06:
Cannot reproduce it. Please open your browser console with F12 and look for errors.

hanipouspilot, 2019-11-06:
Failed to load resource: the server https://[fdfc:ef25:1ba5::1]/cgi-bin/cgi-upload?1573047203685 responded with a status of 405 (Not Allowed)

Uncaught (in promise) SyntaxError: https://[fdfc:ef25:1ba5::1]/luci-static/resources/luci.js?v=git-19.310.44720-c2be304 Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at ClassConstructor.json (https://[fdfc:ef25:1ba5::1]/luci-static/resources/luci.js?v=git-19.310.44720-c2be304:25)
    at eval (https://[fdfc:ef25:1ba5::1]/luci-static/resources/ui.js?v=git-19.310.44720-c2be304:229)

hanipouspilot, 2019-11-06:
Tried both FF and Chrome. Same result.

jow-, 2019-11-06:
Is this using uhttpd or an other webserver?

hanipouspilot, 2019-11-06:
Oops. I forgot to mention. I am testing nginx on this specific device.

jow-, 2019-11-06:
I think nginx support still requires backporting of the latest master fixes. @Ansuel - do you have an overview over the nginx / fastcgi commits that we need to pick into openwrt-19.07 for cgi-io to work properly?

Ansuel, 2019-11-06:
This 3 should fix the problem.

https://github.com/openwrt/packages/pull/10173/files
https://github.com/openwrt/packages/pull/10193
https://github.com/openwrt/packages/pull/10263
<https://github.com/openwrt/packages/pull/10263/files>

hanipouspilot, 2019-11-06:
I cherry-picked those, resolved PKG_RELEASE and HASH conflicts, but still same problem. @Ansuel

Ansuel, 2019-11-06:
Check if the config file got actually updated

hanipouspilot, 2019-11-06:
Which one? /etc/nginx/luci-uwsgi.conf did change.

hanipouspilot, 2019-11-06:
I missed the vassals commit.

hanipouspilot, 2019-11-06:
Now I am getting "bad gateway". I am not familiar with all this web stuff.

Ansuel, 2019-11-06:
Did u restart both uwsgi and nginx?

hanipouspilot, 2019-11-06:
I flashed a new image. So everything was from scratch. I give up for now. Switched to uhttpd for a while.

EXSERENS, 2019-11-08:
select an upgrade file after upload, in browser console:
```
luci.js?v=git-19.311.41964-489fc4b:59 POST https://xxxxxxxxxx/cgi-bin/cgi-upload?1573176670473 413
(anonymous) @ luci.js?v=git-19.311.41964-489fc4b:59
request @ luci.js?v=git-19.311.41964-489fc4b:40
post @ luci.js?v=git-19.311.41964-489fc4b:64
click @ ui.js?v=git-19.311.41964-489fc4b:228

luci.js?v=git-19.311.41964-489fc4b:25 Uncaught (in promise) SyntaxError: Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at ClassConstructor.json (luci.js?v=git-19.311.41964-489fc4b:25)
    at eval (ui.js?v=git-19.311.41964-489fc4b:229)
json @ luci.js?v=git-19.311.41964-489fc4b:25
eval @ ui.js?v=git-19.311.41964-489fc4b:229
Promise.then (async)
click @ ui.js?v=git-19.311.41964-489fc4b:229
```

jow-, 2020-02-23:
Assuming this works now

closed 2020-02-23 by jow-