openwrt/luci #3275: luci-app-sqm 19.07rc1: can't open SQM configuration page through luci menu

svmhdvn, 2019-11-07:
I'm getting the following stacktrace error when trying to open the luci-app-sqm configuration panel:

```
Failed to execute cbi dispatcher target for entry '/admin/network/sqm'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:938: module 'luci.cbi' not found:
	no field package.preload['luci.cbi']
	no file './luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi/init.lua'
	no file '/usr/lib/lua/luci/cbi.lua'
	no file '/usr/lib/lua/luci/cbi/init.lua'
	no file './luci/cbi.so'
	no file '/usr/lib/lua/luci/cbi.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:938: in function </usr/lib/lua/luci/dispatcher.lua:937>
```

on an archer c7 v2 ath79, openwrt 19.07rc1

Snotmann, 2019-11-07:
Same here with WRT1043N V2 and V5

jow-, 2019-11-07:
https://github.com/openwrt/packages/issues/10492