openwrt/luci #3276: luci : realtime graphs show 0 outgoing traffic on bridge and vlan interfaces on traffic menu

Snotmann, 2019-11-07:
Hey guys,

in the realtime graphs seems to be a problem. The download traffic on interfaces is shown correctly, the upload shows 0 all the time:

Inbound:
6.05 Kbit/s
(775 B/s)
Average:
112.53 Kbit/s
(14.07 KB/s)
Peak:
751.52 Kbit/s
(93.94 KB/s)
Outbound:
0 bit/s
(0 B/s)
Average:
0 bit/s
(0 B/s)
Peak:
0 bit/s
(0 B/s)

feckert, 2019-11-08:
Please add a version information (branch)

Snotmann, 2019-11-08:
OpenWrt 19.07.0-rc1, r10649-c4fdb377a2

jow-, 2019-11-08:
I suspect numeric overflow. 

@Snotmann - can you please provide the output of `ubus call luci getRealtimeStats '{ "mode": "interface", "device": "br-lan" }'; sleep 10; ubus call luci getRealtimeStats '{ "mode": "i
nterface", "device": "br-lan" }'`

Replace "br-lan" with one of the affected interfaces.

Snotmann, 2019-11-08:
{
        "result": [
                [
                        1573160963,
                        4186533,
                        46191,
                        91454828,
                        74639
                ],
                [
                        1573160965,
                        4212295,
                        46456,
                        91798417,
                        74945
                ],
                [
                        1573160967,
                        4228606,
                        46570,
                        91965549,
                        75090
                ],
                [
                        1573160969,
                        4229470,
                        46582,
                        91967042,
                        75100
                ],
                [
                        1573160974,
                        4286034,
                        47145,
                        92940340,
                        75883
                ],
                [
                        1573160978,
                        4363771,
                        48245,
                        95550959,
                        77790
                ],
                [
                        1573160982,
                        4432833,
                        48909,
                        97193565,
                        79031
                ],
                [
                        1573161095,
                        5340628,
                        61373,
                        126692901,
                        100990
                ],
                [
                        1573161097,
                        5345641,
                        61455,
                        126956142,
                        101180
                ],
                [
                        1573161098,
                        5351265,
                        61549,
                        127170453,
[... 753 lines cut ...]
                        506150
                ],
                [
                        1573203792,
                        39387480,
                        388129,
                        644969074,
                        506150
                ],
                [
                        1573203794,
                        39387480,
                        388129,
                        644969074,
                        506150
                ],
                [
                        1573203796,
                        39387530,
                        388130,
                        644969074,
                        506150
                ]
        ]
}

jow-, 2019-11-13:
Should have been fixed by https://github.com/openwrt/openwrt/commit/aa89bdcd04676b21cbe3e8a7b8df3545a86d947a

closed 2019-11-13 by jow-