openwrt/luci #3281: Mounted file systems showing wrong info in 19.07.0-rc1

CaboBeach, 2019-11-09:
My attached drive is 3tb with about 1.5tb used, but in Luci>System>Mount Points>Mounted File Systems it's showing as 2.0gb with 0% used, it doesn't appear to affect functionality in any way, so i'm just assuming it's a Luci issue, apologies if it should be reported elsewhere - 

[image]

And i'm not even sure if this is a bug, so i didn't want to open another issue, but i'm getting two ipv6 info boxes in Status>Overview>Network, i can only rememeber getting one in the past, they both show the same info - 

[image]

feckert, 2019-11-10:
What is the output of the command ```/sbin/block info```

CaboBeach, 2019-11-10:
Here's the output - 

```
/sbin/block info
/dev/mtdblock4: UUID="1598287620" VERSION="1" TYPE="ubi"
/dev/mtdblock6: UUID="429114593" VERSION="1" TYPE="ubi"
/dev/ubiblock0_0: UUID="a2304630-82fab1ac-d2ec9479-e8dcb919" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/ubi0_1: UUID="0adae7ef-7afd-4dc3-8e04-b6defd017564" VERSION="w4r0" MOUNT="/overlay" TYPE="ubifs"
/dev/sda1: UUID="0000001800000030" MOUNT="/mnt/storage" TYPE="ntfs"
```

hnyman, 2019-11-10:
Pure guess is some 32-bit integer overflow in the display logic.
(2^32 kB   = 4 TB, so 2 TB would be unsigned 32-bit count of kB.)

jow-, 2019-11-10:
Fixed by https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=aa89bdcd04676b21cbe3e8a7b8df3545a86d947a

closed 2019-11-10 by jow-