openwrt/luci #3282: Status > Overview > Network showing two ipv6 info boxes in 19.07.0-rc1

CaboBeach, 2019-11-09:
I'm not sure if this is a bug, but i've only seen one ipv6 info box in this section in the past, the two that are showing in 19.07.0-rc1 both show the same info -

[image]

jeolives, 2019-11-09:
Looks like you're using luci-theme-material. The introduction of https://github.com/openwrt/luci/commit/6ee61230616245baac02bba903266c881d216008 probably did this. https://github.com/openwrt/luci/commit/1116575513d4874693f56453034e25edcbb4ce1f also needs to be adopted to that theme.

Specifically:
```
.network-status-table .ifacebox-body .ifacebadge > span {
	overflow: hidden;
	text-overflow: ellipsis;
```

CaboBeach, 2019-11-09:
Thanks @jeolives , yes, i'm using luci-theme-material, sorry to ask, but would you happen to know what file 1116575 should be applied to, `find / -name` gives these results for cascade.css - 

```
/overlay/upper/www/luci-static/rosy/cascade.css
/overlay/upper/www/luci-static/material/cascade.css
find: /proc/9500: No such file or directory
/rom/www/luci-static/bootstrap/cascade.css
/www/luci-static/bootstrap/cascade.css
/www/luci-static/rosy/cascade.css
/www/luci-static/material/cascade.css
```

I assumed it would be `/www/luci-static/bootstrap/cascade.css` but looking at it just gives a wall of text and i'm not sure how to find the lines i need if that's the right file.

hnyman, 2019-11-09:
> what file 1116575 should be applied to,  ... I assumed it would be /www/luci-static/bootstrap/cascade.css

Why would you patch the css file for the already fixed "bootstrap" theme, when you aim to fix Material theme?
You need to fix /www/luci-static/material/cascade.css

(Regarding your files found,  /rom is really the ROM (the flashed firmware) and the /overlay contains all later changes.  Just patch /www/luci-static/material/cascade.css and those changes will also be visible as /overlay/www/luci-static/material/cascade.css )

jow-, 2019-11-09:
Already fixed with https://github.com/openwrt/luci/commit/a8a76978298ba1cfddadbb48da73fa2abde82692, will be part of rc2.

closed 2019-11-09 by jow-