openwrt/luci #3288: luci-mod-admin-full: sysupgrade via Luci does not proceed if gzip-ing config files takes too long

fantom-x, 2019-11-10:
19.07-SNAPSHOT
Firefox 70.0.1

Not sure if this is desired behaviour or if it used to work before or not.

It looks like if the config backup takes over 30 seconds, the sysupgrade via LuCI just terminates midway.

sysupgrade from command line works fine. Sysupgrade via Luci used to work for me until a few days ago. Now the new image is uploaded and I can see "gzip -f" running (for a little over 30 seconds). I see some sysupgrade files created under "/tmp/", but the upgrade does not actually proceed beyond creating the config backup.

jow-, 2019-11-10:
How long does the operation take on the CLI?

fantom-x, 2019-11-10:
37 to 40 seconds

fantom-x, 2019-11-10:
All that time is consumed by gzip'ing the config files: I run with "-v" to make sure.

dwardor, 2025-04-15:
Hello,

On 24.10.0 I'm seeing what I think is a similar behavior side effect when backing up config in luci.

It seems the backup.tar.gz is not finished (because of 50 Mb+ of crowdsec mmdb files in /root/.crowdsec that are also backed up) when the webrowser download starts (as if there was a too short timer between start of backup and start of download). So the download finishes before the backuck is finished and the dowloaded file is corrupt because only partial...

Can the download be started only once the backup.tar.gz file is ready ?