openwrt/luci #3293: INT_MAX issue width luci-lib-jsonc

lisaac, 2019-11-13:
Form luci-lib-json switch to luci-lib-jsonc, I found that if number_in_json > INT_MAX, it will return the INT_MAX instead the real value.
```lua
json = require "luci.jsonc"
test ='{"big_number":2147483649}'
-- 2147483649 is INT_MAX + 2
res = json.parse(test)
print(res.big_number)
-- got: 2147483647
```

feckert, 2019-11-13:
Which branch?
What shall he give back according to your opinion?

lisaac, 2019-11-13:
im using 18.06.4, and it should return the real value (2147483649), not the INT_MAX(2147483647)

closed 2019-11-13 by jow-

lisaac, 2020-03-21:
@jow- 
Is there any possible to support int64 in luci.jsonc.stringify ?

```
print(luci.jsonc.stringify({int64=3000000000}))
-- got: {"int64":3000000000.0}
-- This is not an INT
```
I'm using luci-lib-json it works well:
```
print(luci.json.encode({int64=3000000000}))
-- {"int64":3000000000}
```

Thanks

jck112, 2022-04-07:
Can someone take a look at https://github.com/openwrt/luci/pull/5730? I'm trying to fix some of the issues upstream for aarch64 devices that manufacturers (like FriendlyElec) are currently patching in their builds.

commit 8da8f38b (2019-11-13): luci-lib-jsonc: store large JSON integer values as Lua doubles

Fixes: #3293
commit bd1e6d65 (2019-11-13): luci-lib-jsonc: store large JSON integer values as Lua doubles

Fixes: #3293

(cherry picked from commit 8da8f38bf97c925f6f113d3b4793edb23e7e0dc2)
commit ef819b50 (2019-11-13): luci-lib-jsonc: store large JSON integer values as Lua doubles

Fixes: #3293

(cherry picked from commit 8da8f38bf97c925f6f113d3b4793edb23e7e0dc2)
PR #5730 (2022-04-07): luci-lib-jsonc: stringify int use json_object_new_int64 instead
Fixes: #3293