openwrt/luci #3295: [19.07]luci shows incorrect swap info when  a physical swap partition is mounted.

prowong, 2019-11-13:
[image]

jow-, 2019-11-13:
Is the swap larger than 2GB ?

prowong, 2019-11-13:
> 
> 
> Is the swap larger than 2GB ? Please check whether the issue persists after `opkg update; opkg upgrade rpcd; reboot`.

Yes,rpcd is installed, and my platform is x86_64.

jow-, 2019-11-13:
Yes, rpcd is always installed by default, but did you *upgrade* it to the latest version?

prowong, 2019-11-13:
> 
> 
> Yes, rpcd is always installed by default, but did you _upgrade_ it to the latest version?

I compiled the latest version. It happend after the luci-compat was added.

jow-, 2019-11-13:
Please provide the output of `opkg list-installed rpcd` and `ubus call system info`

prowong, 2019-11-14:
OpenWrt 19.07-SNAPSHOT, r10675-5e1864d
 -----------------------------------------------------
root@OpenWrt:~# opkg list-installed rpcd
rpcd - 2019-11-10-77ad0de0-1
root@OpenWrt:~# ubus call system info
{
	"localtime": 1573692247,
	"uptime": 161,
	"load": [
		2816,
		1856,
		160
	],
	"memory": {
		"total": 511537152,
		"free": 362319872,
		"shared": 733184,
		"buffered": 20283392
	},
	"swap": {
		"total": 1327489024,
		"free": 1327489024
	}
}

jow-, 2019-11-14:
This looks like a Kernel level bug. I replicated the problem on a Xiaomi Mir3G by attaching an USB thumb drive and allocating 4GB of swap space. Neither the ubus call, nor the low level `sysinfo(2)` call, no `/proc/swaps` or `free -m` display the correct values. They all cap to ~2GB.

commit 80c9b6fe (2019-11-14): luci-mod-status: fix free swap calculation

Fixes: #3295
commit e82e3d8f (2019-11-14): luci-mod-status: fix free swap calculation

Fixes: #3295

(cherry picked from commit 80c9b6fee0aff23b4c56563612f70ef2c412b7ac)
closed 2019-11-14 by jow-