openwrt/luci #3299: luci-proto-modemmanager Eror openwrt 19.07.0-rc1

predators46, 2019-11-13:
` * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-modemmanager:
 * modemmanager
 * opkg_install_cmd: Cannot install package luci-proto-modemmanager.
 * opkg_install_cmd: Cannot install package modemmanager.
make[2]: *** [Makefile:157: package_install] Error 255
make[1]: *** [Makefile:118: _call_image] Error 2
make: *** [Makefile:197: image] Error 2`

feckert, 2019-11-13:
Same as in https://github.com/openwrt/luci/issues/3294 should be set to Broken

closed 2019-11-13 by feckert

predators46, 2019-11-13:
`debian@linux:~/linux/openwrt$ ./scripts/feeds install modemmanager                     WARNING: Makefile 'package/feeds/luci/luci-app-cifsd/Makefile' has a dependency on 'cifsd-tools', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-proto-modemmanager/Makefile' has a dependency on 'modemmanager', which does not exist
WARNING: No feed for package 'modemmanager' found
debian@linux:~/linux/openwrt$`

jow-, 2019-11-13:
Harmless.

commit ccd7dd52 (2019-11-13): luci-proto-modemmanager: mark as broken

Fixes: #3299