openwrt/luci #3301: luci-app-statistics: load view is wrong

feckert, 2019-11-13:
During my tests and developing with the collectd and so luci-app-statistics I noticed that the values displayed in the Y-axis are misleading, because probably the individual values are added together in each case.

[image]

feckert, 2019-11-13:
@hnyman Do you see it that way?

jow-, 2019-11-13:
Well, the values are stacked. You can disable that by adding `overlay = false` to the options.

feckert, 2019-11-14:
Thanks for the feedback.
I am not sure what the problem is, but your suggestion does not work!
I replaced the following lines

> load__shortterm = { color = "ff0000", title = "1 minute", noarea = true },
> load__midterm   = { color = "ff6600", title = "5 minutes", noarea = true },
> load__longterm  = { color = "ffaa00", title = "15 minutes", noarea = true }

in
`application/luci-app-statistics/luasrc/statistics/rrdtool/definitions/load.lua`
with

> load__shortterm = { color = "ff0000", title = "1 minute", noarea = true, overlay = false },
> load__midterm   = { color = "ff6600", title = "5 minutes", noarea = true,overlay = false },
> load__longterm  = { color = "ffaa00", title = "15 minutes", noarea = true, overlay = false }		

but nothing changed. Does I miss something?

Would it not make more sense to split them up into 3 separate diagramms?
To remove the overlap and display the correct value?

jow-, 2019-11-14:
I might have had the overlay values backwards, try true instead of false.

feckert, 2019-11-14:
Thanks, that's it. I could have thought of that myself. Damn it! :1st_place_medal: 

I will push a fix.
I do also remove the **noarea** so it looks like.

[image]

jow-, 2019-11-14:
I would keep noarea for the 1 minute load and make it the foremost line, so that it draws above the 5 and 15 minute load areas. Since the 1 minute one is the most important imho it should not be covered by the others.

feckert, 2019-11-14:
After merge of #3310 the load view looks like 
[image]

closed 2019-11-14 by feckert

hnyman, 2019-11-16:
@feckert 
Thanks for looking into this. The new look is nice ;-)

PR #3310 (2019-11-14): luci-app-statistics: update misleading load view
Fixes: #3301

During my tests and developing with the collectd and so
luci-app-statistics I noticed that the values displayed in the Y-axis
are misleading, because probably the individual values are added
together in each case. So the view is not corrected.
commit b50974a2 (2019-11-14): luci-app-statistics: update misleading load view

Fixes: #3301

During my tests and developing with the collectd and so
luci-app-statistics I noticed that the values displayed in the Y-axis
are misleading, because probably the individual values are added
together in each case. So the view is not corrected.
commit 039ef1f4 (2019-11-30): luci-app-statistics: update misleading load view

Fixes: #3301

During my tests and developing with the collectd and so
luci-app-statistics I noticed that the values displayed in the Y-axis
are misleading, because probably the individual values are added
together in each case. So the view is not corrected.

(cherry picked from commit b50974a20c9dac0d6b8b479d2e6b54b3c5703ab6)