openwrt/luci #3303: luci-app-statistics: CPU plugin does not display information by CPU

fantom-x, 2019-11-13:
19.07-SNAPSHOT

Here is my config

[image]

And the chart displays combined usage instead of "By CPU"

[image]

feckert, 2019-11-14:
Could you post your generated collectd and uci config?

fantom-x, 2019-11-14:
From luci_statistics 

config statistics 'collectd_cpu'          
        option enable '1'

fantom-x, 2019-11-14:
Hah, so once I save it I get these and then two more charts (one per CPU show up). They are empty at this time.I also see "cpu-0" and "cpu-1" folders under the red directory.

config statistics 'collectd_cpu'          
        option enable '1'                               
        option ReportByCpu '1'            
        option ReportByState '1'          
        option ValuesPercentage '0'

feckert, 2019-11-14:
Ok is this now solved?

feckert, 2019-11-14:
Assume this works as expected.
If not, feel free to reopen this issue.

fantom-x, 2019-11-14:
It is gonna be broken on a fresh install again: two configuration parameters are missing in luci_statistics and collectd is not collecting the relevant data. Luci assumes the wrong defaults for the missing values.
It is working for me now, but a greenfield install will be broken until the user saves that page.

fantom-x, 2019-11-14:
I do not see a way to reopen this issues. It is up to you guys.

jow-, 2020-02-23:
Fixed in master

closed 2020-02-23 by jow-