openwrt/luci #3304: luci: Status / Realtime Graphs / Traffic - missing WNA interface and shows double of each wifi interface

fantom-x, 2019-11-13:
19.07-SNAPSHOT

Issue 1.
I am using PPPoE connection and it used to show up on this page, but not any more.

Issue 2.
If I have an WiFi AP called "Guest" and each band has it's own interface name: "guest2" and "guest5", then I will see two tabs "guest2" and two tabs "guest5" on this page.

jow-, 2019-11-13:
Please provide your /etc/config/wireless as well as the output of `opkg list-installed`. 

Also open your browser's debug console using F12 and paste the following code:
`L.network.getDevices().then(function(d) { s = []; console.debug(JSON.stringify(d, function(k, x) { if (L.isObject(x)) { if (s.indexOf(x) >= 0) return null; else s.push(x); } return x })) })`

Please provide its output as well.

closed 2019-11-13 by jow-

fantom-x, 2019-11-13:
Do you still need anything from me?

Does it fix "pppoe-wan" interface?

jow-, 2019-11-13:
No, yes.

commit 304dd37e (2019-11-13): luci-base: network.js: fix getDevices()

 - Don't filter virtual interfaces
 - Skip duplicate wireless devices

Fixes: #3304
commit ebd354d4 (2019-11-13): luci-base: network.js: fix getDevices()

 - Don't filter virtual interfaces
 - Skip duplicate wireless devices

Fixes: #3304

(cherry picked from commit 304dd37e3fb44843466c38b883fa91b463b5cd1c)