openwrt/luci #3308: luci-app-statistics: ICMP drop rate "10000.00 %" (ping plugin)

fantom-x, 2019-11-14:
19.07-SNAPSHOT

That happens when all pings to a target are failing. 

[image]

feckert, 2019-11-14:
I think this has todo with the following pullrequest
https://github.com/openwrt/packages/pull/9677

and commit

https://github.com/openwrt/luci/commit/9def6faa4e3d35242ff4690ced5c87851441e800#diff-eb03a4159d003171324a4a68d718b6ba

needs more investigation

yousong, 2019-11-14:
The two pull request were targeted for master branch of each repo.  But it seems luci openwrt-19.07 branch forked after the luci-app-staticstics change got merged.

yousong, 2019-11-14:
Probably caused by the merge commit 2f61ab6bd5662a4d441d056c83dc164d3b7adefd .

If we plan to preserve validity of data collectd by current openwrt-19.07 collectd, we need to revert the change on luci-app-statistics for current luci openwrt-19.07 branch.  But the switch is already there in master branch and will be part of next release anyway.

I think we should backport #9677 to openwrt-19.07.  Ping @hnyman  and @jow-

jow-, 2019-11-14:
@yousong - fine with me. Can you take care of the collectd backport?

yousong, 2019-11-14:
Will do.

yousong, 2019-11-15:
@fantom-x  collectd-mod-ping_5.9.0-3 should be starting to be available in the repo.  Please consider giving it a try and report back.

fantom-x, 2019-11-16:
Yup, it show 100% now. Thx.

yousong, 2019-11-16:
Thanks for lettig us know

closed 2019-11-16 by yousong