openwrt/luci #3316: luci-app-shadowsocks-libev: servers page broken in latest luci

ysc3839, 2019-11-16:
@jow- @yousong 
The "Remote Servers" page is broken in new version of luci, showing "TypeError Cannot read property 'id' of undefined".
I found it's because `UIDropdown.__init__` is not called, which is caused by `super` function.
https://github.com/openwrt/luci/blob/1df1421221a2e22334668e6c5524fc359f21d32d/modules/luci-base/htdocs/luci-static/resources/luci.js#L267-L291
I don't know why `superContext` is not a local veriable. Because of this, `superContext` remains `LuCI.View` when `UIDropdown.__init__` is called. So `__init__` is not found and not called.

jow-, 2019-11-22:
@ysc3839 - `superContext` is global because a function invoked by `super()` might call `super()` itself too. In order to prevent an infinite recursion, that inner `super()` function call must not start traversing the prototype chain from `this` again, but start with the scope where the outer `super()` function call left off - which is stored in `superContext`.

I found the logic flaw however; it happened when a `super()` function looking for one symbol, lead to the calling of a method also invoking `super()` but for another symbol. In this particular case an outer `super('renderFooter')` was calling an inner `super('__init__')` which messed up the context saved in `superContext`. To solve this issue, I turned `superContext` into dictionary where the prototype scope pointers are stored keyed by symbol we're currently looking up.

This appears to solve the problem but I'd appreciate if you can give it a spin as well.

This is the patch: https://gist.github.com/jow-/1c6b195f98e6db8c08b35388351f095b

yousong, 2019-11-23:
Confirmed working.  No errors anymore and "Import Links" works.

By the way, I think luci-app-shadowsocks-libev now does not depend on luci-compat.  I have run-tested with luci-compat uninstalled and saw no error.  But I do not yet familiarize myself with the current code enough to be sure.

ysc3839, 2019-11-25:
Tested and seems no issue.

yousong, 2019-11-25:
Reopen as the patch is not yet in the repo.

@jow- , it seems the patch cannot handle the case where the same symbol were called by another object of different class type, e.g.  typeAObj.super('__init__') invokes typeBObj.super('__init__')

jow-, 2019-11-25:
@yousong - that is correct. Any idea how to handle it? Unfortunately one cannot use objects pointers as key in JS so we need to find a way to form a unique identity for the symbol/class combination we're looking up.

yousong, 2019-11-25:
> @yousong - that is correct. Any idea how to handle it? Unfortunately one cannot use objects pointers as key in JS so we need to find a way to form a unique identity for the symbol/class combination we're looking up.

How about making `superContext` collection of stacks keyed by symbol name, pushing and popping on entry/exit of `super()`.

It's ephemeral, specific to the call stack initiated by the first obj.super() call.

`superContext` is stored with obj.

jow-, 2019-11-26:
@yousong - thanks for the suggestion. I opted for a different approach in the end which was easier for me to wrap my head around.

closed 2019-11-26 by jow-

yousong, 2019-11-26:
@jow- Likely it's not enough and a stack-like structure is required.

Let's consider the following case.  The 2nd and 3rd `super('render')` calls are at the same level.  Context will be cleared on return from 2nd call, causing the 3rd call starting from scratch invoking method of child class.

```
obj.super('render')
  obj.super('render')
  obj.super('render')
```

commit 374c23cd (2019-11-26): luci-base: luci.js: properly handle nested Class.super() calls

Introduce a unique __id__ attribute per class and use that, together with
the symbol name being looked up to, to form a unique key which is used to
store the prototype scope pointer.

This fixes cases where a call to super() invokes a procedure which is
calling super() as well on a different class or symbol.

Fixes: #3316
commit 3ad8ef4e (2019-11-26): luci-base: luci.js: properly handle nested Class.super() calls

Introduce a unique __id__ attribute per class and use that, together with
the symbol name being looked up to, to form a unique key which is used to
store the prototype scope pointer.

This fixes cases where a call to super() invokes a procedure which is
calling super() as well on a different class or symbol.

Fixes: #3316

(cherry picked from commit 374c23cdab53c5681acafb561977d927c361408a)