openwrt/luci #3319: [bug report] luci-app-ttyd cannot display in luci menu

Mosney, 2019-11-16:
[image]

castillofrancodamian, 2019-11-16:
It is in the "Services" tab.

Mosney, 2019-11-17:
> It is in the "Services" tab.

[image]
services tab look like this.

feckert, 2019-11-17:
Do you have a configuration file (/etc/config/ttyd)?
See code https://github.com/openwrt/luci/blob/2390e7297601767ddc7f25ab5fb606500117ffea/applications/luci-app-ttyd/luasrc/controller/ttyd.lua#L5-L8

ysc3839, 2019-11-17:
Nothing wrong here.
Versions: LuCI Master (git-19.314.67968-e8abe15) / OpenWrt SNAPSHOT r11526-f84d6d2f3a
I guess it's 19.07's issue?

ysc3839, 2019-11-17:
Yes, because ttyd in 19.07 is not updated. luci-app-ttyd should be marked as broken.
https://github.com/openwrt/packages/tree/openwrt-19.07/utils/ttyd

hnyman, 2019-11-17:
> luci-app-ttyd should be marked as broken.

Better might be to backport the changes in the main ttyd app from master to 19.07

jow-, 2019-11-17:
Can anybody take care of either backporting ttyd or - if that is rejected - marking luci-app-ttyd as `@BROKEN` ?

hnyman, 2019-11-17:
The two ttyd commits in master about 1.5.1 and 1.5.2 (and uci config) look so simple that I cherry-picked them into 19.07.

Mosney, 2019-11-18:
> Do you have a configuration file (/etc/config/ttyd)?
> See code
> 
> https://github.com/openwrt/luci/blob/2390e7297601767ddc7f25ab5fb606500117ffea/applications/luci-app-ttyd/luasrc/controller/ttyd.lua#L5-L8

I don't find this file, so it means I need create a conf file?

feckert, 2019-11-18:
I do not know how the file `/etc/config/ttyd` does looks like, but I think you need a default config so that the configuration gets shown. This code section checks if the file exists and only then the menu entry is shown.

Mosney, 2019-11-18:
I think it already been fixed. Just upgrade packages.
[image]

closed 2019-11-18 by Mosney