openwrt/luci #3325: LuCI unable to connect to the internet?

robman501a, 2019-11-17:
Here's my setup:
Main router: ASUS RT-AC1900P (Asuswrt-Merlin)
OpenWRT device 1: Netgear WNDR3700v1 (acting as a switch https://openwrt.org/docs/guide-user/network/wifi/dumbap)
OpenWRT device 2: WD MyBookLive NAS
[image]
The problem: The NAS is unable to access the internet at all (LAN port connected to main routers LAN port, IP address of NAS changed to 192.168.1.3). The WNDR3700v1 can't access the internet at all through its web interface but connected devices do have internet access.

ysc3839, 2019-11-18:
I think it's not a bug about luci. You should post this to https://forum.openwrt.org/
By the way, try `traceroute openwrt.org` or ping your main router.

robman501a, 2019-11-18:
Okay, I'll head over there. Thanks for suggesting that.
By the way, the second router with OpenWRT can ping my main router. :)

closed 2019-11-18 by robman501a