openwrt/luci #3340: Interface IPv6 configuration: PD and static IPv6 address

madires, 2019-11-21:
Hi!

I found an issue with the IPv6 interface configuration when applying an assignment of a PD subprefix and a static IPv6 address in OpenWrt 18.06.5. Typically you get a public IPv6 prefix from your ISP and also want to have a static ULA prefix in your LAN. LuCI allows me to set either the PD assignment or a static IPv6 address. But /etc/config/network supports both at the same time:

config interface 'lan'
        option type 'bridge'
        option ifname 'eth0.1'
        option proto 'static'
        option netmask '255.255.255.0'
        option ip6assign '64'
        option ip6ifaceid '::1'
        option ipaddr '192.168.1.1'
        option ip6addr 'fd00:dead:beef:1::1/64'

The example above works fine. Unfortunately, LuCI displays only the PD configuration and will remove the "option ip6addr" when saving the current configuration. It's annoying to have to add the "option ip6addr" manually again each time. The other workaround is to use an alias interface which is also cumbersome. It would be nice to have both settings (PD and static IPv6 address) at the same time.

Cheers,
 Markus

cvmiller, 2020-01-19:
As a work-around you can set the ULA in the globals section:

```
config globals 'globals'
	option ula_prefix 'fd00:dead:beef::/64'
```

adelton, 2025-01-10:
I can see the
```
o.depends('ip6assign', '');
```
is there since 029631ec489b2d28a9e4b3a6ef4f2de9e999b956, in the original form of
```
ip6addr:depends("ip6assign", "")
```

@sbyx, would you remember the original reasoning? The commit message says "Make static IPv6-configuration more usable" but @madires suggests that being able to have both `ip6addr` and `ip6assign` would actually be more useful.

What would break if we just removed that `o.depends('ip6assign', '');` in https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/protocol/static.js#L187?

systemcrash, 2025-01-10:
I recall looking at this, when I was interested to increase the visibility of (static) IPv6. Both should be configurable.

systemcrash, 2025-01-10:
PD and static should both be possible, unless we can ascertain a reason why that might be undesirable.