openwrt/luci #3343: Cannot edit host config via luci->network->DHCP and DNS->Static leases

Mosney, 2019-11-22:
[image]
As screenshot, I cannot use 'Static leases' section to (1)edit exist dhcp host config (2) create new host. But it is OK to direct edit /etc/config/dhcp file.

**Version**
------
OpenWrt: 19.07rc1 (and opkg source also 19.07)
luci: git-19.324.23079-c0b9fbb-1	
dnsmasq: 2.80-14	

**luci notice in the screenshot**
----
```
TypeError
Cannot read property 'getAttribute' of null
```

**Browser console log**
------
```
validation.js?v=git-19.324.23079-c0b9fbb:57 Uncaught (in promise) TypeError: Cannot read property 'getAttribute' of null
    at ClassConstructor.unique (validation.js?v=git-19.324.23079-c0b9fbb:57)
    at ClassConstructor.apply (validation.js?v=git-19.324.23079-c0b9fbb:6)
    at ClassConstructor.list (validation.js?v=git-19.324.23079-c0b9fbb:53)
    at ClassConstructor.validate (validation.js?v=git-19.324.23079-c0b9fbb:9)
    at ClassConstructor.triggerValidation (ui.js?v=git-19.324.23079-c0b9fbb:4)
    at ClassConstructor.callClassMethod (luci.js?v=git-19.324.23079-c0b9fbb:192)
    at ClassConstructor.triggerValidation (form.js?v=git-19.324.23079-c0b9fbb:115)
    at ClassConstructor.checkDepends (form.js?v=git-19.324.23079-c0b9fbb:80)
    at ClassConstructor.checkDepends (form.js?v=git-19.324.23079-c0b9fbb:56)
    at ClassConstructor.eval (form.js?v=git-19.324.23079-c0b9fbb:52)
unique @ validation.js?v=git-19.324.23079-c0b9fbb:57
apply @ validation.js?v=git-19.324.23079-c0b9fbb:6
list @ validation.js?v=git-19.324.23079-c0b9fbb:53
validate @ validation.js?v=git-19.324.23079-c0b9fbb:9
triggerValidation @ ui.js?v=git-19.324.23079-c0b9fbb:4
callClassMethod @ luci.js?v=git-19.324.23079-c0b9fbb:192
triggerValidation @ form.js?v=git-19.324.23079-c0b9fbb:115
checkDepends @ form.js?v=git-19.324.23079-c0b9fbb:80
checkDepends @ form.js?v=git-19.324.23079-c0b9fbb:56
(anonymous) @ form.js?v=git-19.324.23079-c0b9fbb:52

```

gnurdware, 2019-11-22:
Can confirm this bug ( or a similar one. )

Recently installed latest OpenWRT on TP-Link Archer C7 v2 router, and attempting to add a new static route via LuCI interface causes browser to hang after pressing "save."

OpenWRT 19.07.0-rc1
LuCI openwrt-19.07 branch (git-19.323.27428-ee61723)

To work around it, I installed the "nano" editor, and modified the /etc/config/dhcp configuration file and added the following entry:

config host
   option dns '1'
   option mac 'xx:xx:xx:xx:xx:xx'
   option ip '192.168.xxx.xxx'
   option leasetime '12h'
   option name 'hostname'

After saving the file and exiting nano, hitting refresh in the LuCI browser interface displayed the static entry correctly.

jow-, 2019-11-22:
@gnurdware - your issue is a different one which has been fixed with https://github.com/openwrt/luci/commit/4f0d67a2897d6592cb0f7562fa042dd4987fd7eb.

jow-, 2019-11-22:
Dupe of #3341

closed 2019-11-22 by jow-