openwrt/luci #3350: collectd-mod-cpufreq: missing for ipq40xx target

fantom-x, 2019-11-22:
19.07-SNAPSHOT

Is there a reason why **collectd-mod-cpufreq** is not built for this target?

hnyman, 2019-11-23:
Looks like mismatch between the buildbot options for ipq40xx and the definitions in the collectd Makefile.

The collectd-mod-cpufreq module has been enabled only for those targets that are known to have processors that do frequency scaling.
https://github.com/openwrt/packages/blob/master/utils/collectd/Makefile#L371
`@(TARGET_x86||TARGET_x86_64||TARGET_mvebu||TARGET_ipq806x||TARGET_armvirt))`

ipq806x routers  use armvirt phase1 SDK for the packages phase2 build (arm_cortex-a15_neon-vfpv4), so it gets built for ipq806x R7800 privately (ipq806x) and in the buildbot (armvirt).

Your ipq40xx has been defined as cortex-a7-neon-vfpv4:
https://github.com/openwrt/openwrt/blob/master/target/linux/ipq40xx/Makefile#L7
```
CPU_TYPE:=cortex-a7
CPU_SUBTYPE:=neon-vfpv4
```

cortex-a7-neon-vfpv4 uses brcm2708/bcm2709 SDK in phase2 packages build. And that module is not in the list defined in collectd-mod-cpufreq.
http://buildbot.openwrt.org/master/packages/builders/arm_cortex-a7_neon-vfpv4/builds/93/steps/downloadsdk/logs/stdio

So, the answer is:
It does not get built in buildbot as ipq40xx is defined as cortex-a7-neon-vfpv4 that takes the CPU features from the brcm2708-bcm2709 SDK definitions, and collectd-mod-cpufreq does not know that that CPU family supports scaling.

It should work for your own builds if you add TARGET_ipq40xx to that line in the collectd Makefile, as the packages are built according to the main target, not based on a separate SDK.

The collectd Makefile definition might be changed to include also TARGET_bcrm2708 (????), and the module might then get compiled in buildbot.  (I am not sure if the subtarget should be defined somehow there, as likely not all bcrm2708 chips do support scaling).

Ps. 
collectd is in the packages feed repo, not in this LuCI repo. 
So I close this LuCI issue in any case.

closed 2019-11-23 by hnyman

fantom-x, 2019-11-23:
Shall I open an issue in the packages repo?

hnyman, 2019-11-23:
Have you tested it on your own builds?  And does scaling itself work?
(if it works, the module could be enabled)

fantom-x, 2019-11-23:
The scaling definitely works: I can change the governor, frequencies, etc. Everything is just like in ipq806x.

hnyman, 2019-11-23:
I may push the change enabling the collectd for your target once I figure out the exact correct TARGET. TARGET_ipq40xx would help your personal builds, but buildbot phase2 needs that brcm2708/bcm2709 in the correct format.

(funny, just when I was starting to look for this collectd thing, I got sidetracked with identifying and regression window hunting for https://bugs.openwrt.org/index.php?do=details&task_id=2624 )

fantom-x, 2019-11-23:
quid pro quo :)

I can help you there, because I got that crash an hour ago. I had to revert to the build below just from a few hours back to get online again. I hope that will help.

**openwrt-19.07-snapshot-r10727-e30ca0d90a-ipq806x-generic-netgear_r7800-squashfs-sysupgrade.bin**

hnyman, 2019-11-23:
No, I already made the regression tests:
Good: r10732-b177b180bb
Bad: r10733-436dbf12aa 
https://github.com/openwrt/openwrt/commit/436dbf12aa7381bd6db1b03b36be5a73df386eb4

hnyman, 2019-11-23:
> The scaling definitely works: I can change the governor, frequencies, etc. Everything is just like in ipq806x.

I added cpufreq for ipq40xx as https://github.com/openwrt/packages/commit/5e0d5c9c7fd9d6637af0c3443e495f7492e19460

fantom-x, 2019-11-23:
Do you mind back porting to 19.07? I can only test there.

fantom-x, 2019-11-27:
I can confirm that it is working fine in master on GL-B1300 (ipq400xx).

fantom-x, 2019-11-28:
Would you mind porting this to 19.07?

hnyman, 2019-11-30:
backported as cherry-pick 
https://github.com/openwrt/packages/commit/00803ffc91e80b16e9c1603ff32106d42e255923