openwrt/luci #3351: LuCI: Fails to open OpenWrt Patchworks series 144179

jeolives, 2019-11-23:
Currently not a problem in `master`, but will be if https://patchwork.ozlabs.org/project/openwrt/list/?series=144179 gets merged as is. 

I don't really know how to use the mailing list so I've thought of putting it here as its tangentially related.

[image]

@ynezz

ynezz, 2019-11-23:
@jeolives can you share `/etc/config/luci` ?

jeolives, 2019-11-23:
```
config core 'main'
        option lang 'auto'
        option mediaurlbase '/luci-static/bootstrap'
        option resourcebase '/luci-static/resources'

config extern 'flash_keep'
        option uci '/etc/config/'
        option dropbear '/etc/dropbear/'
        option openvpn '/etc/openvpn/'
        option passwd '/etc/passwd'
        option opkg '/etc/opkg.conf'
        option firewall '/etc/firewall.user'
        option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
        option sessionpath '/tmp/luci-sessions'
        option sessiontime '3600'

config internal 'ccache'
        option enable '1'

config internal 'themes'
        option Bootstrap '/luci-static/bootstrap'

config internal 'apply'
        option rollback '30'
        option holdoff '4'
        option timeout '5'
        option display '1.5'

config internal 'diag'
        option dns 'openwrt.org'
        option ping 'openwrt.org'
        option route 'openwrt.org'

config ifstate
        option interface 'lan'
        option ifname 'eth0.1 radio0.network1 radio1.network1'
        option bridge 'true'
```

jow-, 2019-11-23:
The message above happens if rpcd crashes or is unreachable. Since it is unrelated to LuCI, not caused by LuCI and not going to be fixed in LuCI, I'll close this here.

closed 2019-11-23 by jow-

ynezz, 2019-11-23:
@jeolives thanks for reporting, if you're interested you can find fixed series here https://gitlab.com/ynezz/openwrt-libubox/commits/wip