openwrt/luci #3356: PXE boot menu - Looking for a LuCI module developer

PeterSurda, 2019-11-24:
I've been trying to find a developer to write a LuCI module, sadly I haven't had any feedback so far so I though I'll post it here. The description of the module is [at bountify](https://bountify.co/openwrt-luci-module-for-pxe-boot-menu): it is for making the default boot option in pxelinux configurable per host through the web interface, so that the administrator doesn't have to edit the files in a shell. I think that for someone familiar with LuCI it should be easy. I don't have time to familiarise myself with the framework so I'm offering a bounty instead. The one at bountify is expired but I still have a budget available. The module is going to be open source so if there is interest after it's written, a PR can be made.

Nicarim, 2020-01-22:
@PeterSurda Did you get anyone to do it yet?

PeterSurda, 2020-01-22:
@Nicarim No I haven't found anyone to do it.

ysc3839, 2020-01-22:
I guess people usually make contributions based on their own demands. For example, some months ago I decided to replace Samba with cifds (now smbd) but there's no luci-app-cifsd so I wrote one.
PXE boot usually used in a place with many computers. Most OpenWrt users are home user using OpenWrt on their router, so they don't need PXE. Also PXE users are IT professional who knows how to configure PXE boot manually, they don't need a Web UI.
So finally you either keep on waiting or trying to make one by yourself.

vKnmnn, 2020-03-22:
we should try do get some money together to have this implemented. a sort of bounty. I'd be willing to give a few € for this feature.

We should also define more clearly what it is that we want

systemcrash, 2021-08-31:
What you want in this issue is not what the majority of openwrt users will ever aspire to. Judging by the fact that the [sheer download numbers of binary variants](https://downloads.openwrt.org/stats/awstats.downloads.openwrt.org.allextra1.html) are for models with limited flash memory, and these features will require more substantial chunks of disk-space, most people don't have the luxury of implementing this 'locally', without hacking around with USB drives and mounting those at boot etc. I run a NAS which handles all of the files, but the DHCP server is on WRT, so the PR #5304 improves that part.

adelton, 2025-01-11:
The bountify URL (as well as the whole site in general) currently produces 503 Service Unavailable / Application error. So the description of the desired module no longer seem to exist.

closed 2025-01-11 by systemcrash