openwrt/luci #3357: [luci.js] invalid XML

n8v8R, 2019-11-25:
* exhibits on Master and 19.07
* exhibits with different browsers
___
Accessing _/cgi-bin/luci/admin/status/processes_ prints (depending on browser)

> Failed to set the 'innerHTML' property on 'Element': The provided markup is invalid XML, and therefore cannot be inserted into an XML document.

or

> SyntaxError
An invalid or illegal string was specified
___
Browser console points to _luci.js_

`162 else if(children!==null&&children!==undefined){node.innerHTML=''+children;return node.lastChild;}`

jow-, 2019-11-25:
Which theme?

n8v8R, 2019-11-25:
OpenWrt theme only, does not exhibit with Bootstrap or Material.

luci-mod-status git-19.329.02080-e3b3653-1.0 (that is on Master).

jow-, 2019-11-25:
Hm, there is no revision `e3b3653` in the repo. The `1.0` revision looks odd as well. Is this some LuCI fork?

n8v8R, 2019-11-25:
No fork

`opkg info luci-mod-status`

> Package: luci-mod-status
> Version: git-19.329.02080-e3b3653-1.0
> Depends: libc, luci-base, libiwinfo, libiwinfo-lua
> Status: install user installed
> Section: luci
> Architecture: arm_cortex-a9_vfpv3
> Size: 25498
> Filename: luci-mod-status_git-19.329.02080-e3b3653-1_arm_cortex-a9_vfpv3.ipk
> Description: LuCI Status Pages
> Installed-Time: 1574668515

jow-, 2019-11-25:
That version is newer than the current LuCI branch tip and newer than whats available on the download server. You must be building from some different source, or with custom commits applied or something.

n8v8R, 2019-11-25:
A bit strange since the LuCI packages are being pulled straight from the OpenWrt feed (server).

Closing since being a minor issue and exhibiting only OpenWrt theme.

closed 2019-11-25 by n8v8R