openwrt/luci #3363: "WPA3 Enterprise" / "WPA2/3 Enterprise (mixed)" in Wireless configuration

tiffchros, 2019-11-25:
I'm missing "WPA3 Enterprise" / "WPA2/3 Enterprise (mixed)" in Wireless configuration on Luci.
It is already implemented according to https://openwrt.org/docs/guide-user/network/wifi/basic

Steps to reproduce:

- Install https://downloads.openwrt.org/releases/19.07.0-rc1/targets/ipq806x/generic/openwrt-19.07.0-rc1-ipq806x-generic-tplink_c2600-squashfs-sysupgrade.bin on Archer C2600, disabled "Keep settings and retain the current configuration"
- Set password via Luci
- Connect via SSH:
  - root@OpenWrt:~# opkg update
  - root@OpenWrt:~# opkg remove wpad-basic
  - root@OpenWrt:~# opkg install wpad-openssl
  - root@OpenWrt:~# reboot
- Login via Browser
- Goto Network -> Wireless -> [Edit] on 2,4 GHz or 5GHz interface
- Interface Configuration -> Wireless security

Option is only:
- WPA2-PSK (strong security)
- WPA2-EAP (strong security)
- WPA3-SAE (strong security)
- WPA2-PSK/WPA3-SAE Mixed Mode (strong security)
- WPA-PSK/WPA2-PSK Mixed Mode (medium security)
- WPA-PSK (medium security)
- WPA-EAP (medium security)
- WEP Open System (weak security)
- WEP Shared Key (weak security)
- OWE (open network)
- No Encryption (open network)

... and unfortunately no
- WPA3-EAP / WPA2-EAP Mixed Mode
- WPA3-EAP

there.
I also tried wpad-wolfssl, but the same issue.

abraxxa, 2020-01-11:
I have the same problem with the 19.07.0 release version and haven't found a solution so far.

TimelessNL, 2020-01-12:
Not that it is a very good choice but "WPA-EAP/WPA2-EAP Mixed Mode (medium security)" is not listed either. Maybe related why mixed modes do not support EAP?

TimelessNL, 2020-01-12:
I just tried something, it just seems like a config GUI problem. when I manually set `option encryption 'wpa3-mixed'` LuCI will correctly display `mixed WPA2/WPA3 802.1X (CCMP)` in the overview, but when changing settings it will display random settings and mode is set to `WPA2-PSK` and no EAP settings at all.

urbenlegend, 2020-01-20:
I am getting the same issue here as well. I see WPA2-EAP, but not WPA3-EAP.

Robby-, 2020-01-26:
The same issue exists in the master branch.

abraxxa, 2020-03-07:
Issue still exists in 19.07.2.

stringtrim, 2020-03-26:
Issue still exists in OpenWrt 19.07.2 r10947-65030d81f3

commit cae80453 (2020-03-26): luci-mod-network: add WPA3-EAP and mixed WPA2/WPA3-EAP settings

Fixes: #3363
commit f1f33c8a (2020-03-26): luci-mod-network: add WPA3-EAP and mixed WPA2/WPA3-EAP settings

Fixes: #3363

(backported from commit cae804536785cac5d5a2b3b8e3b9d38c1e623148)
closed 2020-03-26 by jow-