openwrt/luci #3369: "bad regex: Out of memory" in logs

zorun, 2019-11-27:
I am running OpenWrt 18.06.5 on a Netgear DM200 VDSL modem.

After logging in LuCI, I saw this in `logread`:

```
19:33:03 2019 daemon.err uhttpd[936]: luci: accepted login on / for root from XX.XX.XX.XX
19:33:39 2019 daemon.err uhttpd[936]: expr: bad regex '.*nFEC=\([-\.[:alnum:]]*\).*': Out of memory
```

The device has 64 MB of RAM, so this is a bit surprising, especially because the regex does not look that complex. I'm not sure which part of LuCI triggered this ("FEC" means it is probably related to the DSL status on the home page)

Unfortunately I couldn't reproduce it more than once.

jow-, 2019-11-28:
That looks as if it is caused by the dsl_control init script which is periodically invoked. LuCI itself does not contain such a regex

jow-, 2019-11-29:
Closing this as it is not related to LuCI

closed 2019-11-29 by jow-