openwrt/luci #3371: luci-app-statistics - Graphs fail with permission denied - 19.07-RC1

DougBarton, 2019-11-28:
I installed the package via the instructions [here](https://openwrt.org/docs/guide-user/luci/luci_app_statistics). I can configure the collection of data in luci, but when I click the pulldown for graphs I get this:

```Failed to execute template dispatcher target for entry '/admin/statistics/graph'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:74: Failed to load template 'admin_statistics/index'.
Error while parsing template '/usr/lib/lua/luci/view/admin_statistics/index.htm':
Permission denied
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:74: in function '__init__'
	/usr/lib/lua/luci/util.lua:65: in function 'Template'
	/usr/lib/lua/luci/template.lua:27: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:920: in function </usr/lib/lua/luci/dispatcher.lua:919>
```

It may be relevant that I'm using an imagebuilder image that I built in the packages for. As far back as I can remember (at least 10 years) when upgrading using imagebuilder I have had to go in on the command line and do `chmod -R go+rX /www` in order to get luci to load, otherwise it gives a permission denied error as well. I have tried doing a similar change to the directories related to luci-app-statistics, and it didn't help. 

I'm using 19.07-RC1, I can try the latest release if someone can confirm that will help. Everything else works for me, and this is the first time I'm trying the statistics package. 

My router is a TP-Link Archer C7 v2, AC1750, and I'm using the openwrt-imagebuilder-19.07.0-rc1-ar71xx-generic.Linux-x86_64.tar.xz package.

jow-, 2019-11-28:
Check if your rrd file directory and any parent directories leading to it are readable by “nobody”.

DougBarton, 2019-11-28:
> Check if your rrd file directory and any parent directories leading to it are readable by “nobody”.

They are, and the files are 0644 as well. But thank you for the response.  :)

DougBarton, 2019-11-29:
I tried again using 18.06.5 with imagebuilder, both with and without incorporating the luci-app-statistics packages, and neither worked. 

What did work is using a downloaded image and adding luci-app-statistics with opkg. 

So aside from the /www permissions problem with imagebuilder images that's been around for a decade or more, there is a larger problem with other parts of the system as well. 

So I'm going to edit the subject of this bug, and hopefully it can get some attention. :)

jow-, 2019-11-29:
What is your host system umask?

DougBarton, 2019-11-29:
> What is your host system umask?

0022, but I'm not sure that should matter. Shouldn't part of the install process be to make sure that the permissions are set correctly?

jow-, 2019-11-29:
*shrug* - in any case this does not look like a LuCI specific issue anymore.

DougBarton, 2019-11-29:
ok, fair enough, so where should it be posted, and how do I move it?

jow-, 2019-11-29:
Maybe bugs.openwrt.org - but I'm afraid it will not get much traction there either. I gave it a very quick test run and was unable to reproduce your problem using the x86/64 imagebuilder.

Your initial quote:
> As far back as I can remember (at least 10 years) when upgrading using imagebuilder I have had to go in on the command line and do chmod -R go+rX /www in order to get luci to load, otherwise it gives a permission denied error as well.

Sounds very odd to me. If that'd be a widespread problem (LuCI not working on images built with IB) - I'm sure there would have been way more reports, especially if it has been broken for 10 years straight. I suspect something in your environment is broken, or maybe you embed third party packages or a files/ overlay that introduces improper permissions.

DougBarton, 2019-11-29:
I'm not doing anything fancy like that, just `make image PROFILE=archer-c7-v2 PACKAGES=....`

I'll ask on the forum, thanks for the responses.  :)

closed 2019-11-29 by DougBarton