openwrt/luci #3378: [RFC]Require JavaScript for new apps?

aparcar, 2019-12-01:
I'm wondering if all new LuCI apps should be written in JavaScript and LUA is from now on considered legacy and not accepted for new apps anymore.
The documentation seem straight forward enough to easily convert existing LUA code to JS.

Please share your thoughts.

hnyman, 2019-12-01:
I feel that it might be too strict to "require" JavaScript, but it might be "preferred".

Personally I think that it does not make much difference in simple uci config apps without much calculation, logic or interactive status screens. The goal of JS is to move the CPU intensive processing to the client side, but many apps are rather simple and just used for basic uci config.

aparcar, 2019-12-01:
Maybe I'm wrong but I though another advantage is to get rid of the lua
runtime and save some storage. Eventually lowering also the languages to
know for maintenance.

jow-, 2019-12-02:
Yes, the goal is to get rid of Lua completely for the LuCI base.

aparcar, 2019-12-02:
Should we pin this?

howard0su, 2020-03-01:
Where is migration guide to migrate from LUA-based to JS-based project?

oxr463, 2020-05-20:
So the JavaScript apps run on the client-side in the browser vs Lua running via cgi-bin on the device, right?

feckert, 2020-12-22:
Unfortunately, I saw this too late and did not think on this, but I have merged this https://github.com/openwrt/luci/pull/4664 and it is written in lua.
I have now another [P/R](https://github.com/openwrt/luci/pull/4665) which is also lua and which I also just wanted to merge.
So that I don't make the same mistake in the review and tell people right away that we want javascript, we should communicate this clearly and be in agreement.

I can not revert this https://github.com/openwrt/luci/pull/4665 but I can request to write this https://github.com/openwrt/luci/pull/4664 in javascript or should I merge them?

PolynomialDivision, 2021-01-15:
> write this #4664 in javascript or should I merge them?

I can rewrite that in javascript, but can u maybe link to a good example?
Maybe luci-mod-status is a good example?

aparcar, 2021-01-15:
@PolynomialDivision @howard0su 

I'm afraid at this point the best documentation for migrating from Lua is the [extensive LuCI JS documentation](http://openwrt.github.io/luci/jsapi/index.html) in combination with pull requests that did the migration in the past, there are multiple.

@PolynomialDivision if you go through the process and find the time it would be great to share the experience and write it down in a document.

PolynomialDivision, 2021-01-16:
> @PolynomialDivision if you go through the process and find the time it would be great to share the experience and write it down in a document.

Okay. :) Actually, I have to do document that since we have to rewrite all our luci apps for our community wireless network. Maybe @JonnyTischbein will help me? ;)

JonnyTischbein, 2021-01-16:
> Okay. :) Actually, I have to do document that since we have to rewrite all our luci apps for our community wireless network. Maybe @JonnyTischbein will help me? ;)
I would love to! :)

PolynomialDivision, 2021-02-08:
I tried now rewrite the luci-app-babeld in javascript:
https://github.com/openwrt/luci/pull/4780

JonnyTischbein, 2021-02-10:
I did a new rewrite of @PolynomialDivision pr https://github.com/openwrt/luci/pull/4791 with javascript dom.

As @aparcar asked I would like to write down my process, but I'm not sure whether using the javascript dom instead of luci dom is okay ?
Also I'm unsure whether using the div classed table as requested here https://github.com/openwrt/luci/pull/3927#issuecomment-615781467 is implemented in right way.

Could you please give me some feedback ?

stangri, 2021-02-15:
I was looking to start converting my luci apps and have some questions, should I ask them here or create a new migration-related issue?

aparcar, 2021-02-15:
@stangri please create single issues for that, I think @PolynomialDivision and @JonnyTischbein are a tag team to tackle any upcoming questions.

pmelange, 2021-07-20:
It would be very helpful if luci-app-splash were ported to JS as an example of an app which has both public and also non-public (authorization requited) content.  @PolynomialDivision @JonnyTischbein, could you imagine doing this?

JonnyTischbein, 2021-07-28:
@pmelange in the new weeks I won't be able to do that. But I am interested. :/

pmelange, 2021-09-07:
> @pmelange in the next few weeks I won't be able to do that. But I am interested. :/

Is there any progress on this?

bam80, 2022-09-26:
ping?

bam80, 2022-09-27:
Sorry for resurrecting the old topic.
I'll just leave this here as it wasn't apparently mentioned:
https://weimarnetz.de/blog/artikel/howto-write-new-luci-apps
> With this example app it will be easier to develop new or migrate older LuCI apps. So we hope we can get rid of luci-lua dependencies to create smaller images and have faster routers.

akhamidullin1, 2023-11-17:
Hello everyone.
Thank you very much for the conversation and provided links.
Got able to start redesigning luci app from [lua](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs) based solution to [new](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs/-/tree/2023) build system.
Currently i stuck on ability to implement re-render certain widgets of UI that was implemented by [window.requestAnimationFrame](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs/-/blob/master/luci/view/gs.htm#L361) and [setInterval](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs/-/blob/master/luci/view/gs.htm#L365) functions. Could someone suggests way to implement similar functionality please? Or maybe it is possible to use that functions in new build system somehow? Tried to [look](https://github.com/search?q=repo%3Aopenwrt%2Fluci+requestAnimationFrame&type=code) through the code, - looks no one uses **window.requestAnimationFrame** in new build system.

wireless-road, 2023-11-17:
Hello everyone.
Thank you very much for the conversation and provided links.
Got able to start redesigning luci app from [lua](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs) based solution to [new](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs/-/tree/2023) build system.
Currently i stuck on ability to implement re-render certain widgets of UI that was implemented by [window.requestAnimationFrame](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs/-/blob/master/luci/view/gs.htm#L361) and [setInterval](https://git.m2m-tele.com/openwrt-packages/gs/luci-gs/-/blob/master/luci/view/gs.htm#L365) functions. Could someone suggests way to implement similar functionality please? Or maybe it is possible to use that functions in new build system somehow? Tried to [look](https://github.com/search?q=repo%3Aopenwrt%2Fluci+requestAnimationFrame&type=code) through the code, - looks no one uses **window.requestAnimationFrame** in new build system.

sourabh-burse, 2024-02-10:
Hello guys,
This question may be irrelevant to this thread, sorry for that.

So I have a lua based luci app that I am trying to convert to js, slowly progressing on that but in meantime I want to fix a major issue with my lua based luci app.

Issue is: I am loading values in the section from model/cbi and there are basically 3 sections which are connected to each other with a unique id, when there are 100+ entries in the config file my luci app takes forever to load.

I know that I should be reading the uci file from controller and then pass it to cbi but I am not sure how to do it, and if i do it how to handle data update?