openwrt/luci #3380: httpclient.parse_url does not accept IPv6 link-local

ilario, 2019-12-01:
The `httpclient.parse_url` function included with a54baf7be0d293b50c6b2754c11d61e6e2ad82af does not accept as a valid URI a direction including an IPv6 link-local, which is something like this:
`http://[fe80::c24a:ff:fedd:6800%br-lan]/cgi-bin/shared-state/babeld-hosts`
We use these kind of addresses in [shared-state](https://github.com/libremesh/lime-packages/tree/master/packages/shared-state) tool.

The line which should match the IPv6 is this:
https://github.com/openwrt/luci/blob/a692ca789ae916b1393a39236ff8cf3e8a431717/libs/luci-lib-httpclient/luasrc/httpclient.lua#L108

and it does not like that there is some `%br-lan` before the closing `]`.
Thanks!

ilario, 2019-12-01:
Seems that neither ip.IPv6 likes the IPv6 link-local :/

ilario, 2019-12-04:
I forgot to specify that this is a regression: with httpclient from OpenWrt 18.06 also the IPv6 link-local were working.

jow-, 2019-12-09:
Please try the following patch: https://gist.github.com/jow-/b93d2454a42e224096982eb86221806c

ilario, 2019-12-09:
Thanks!
I tested and I think there are two issues:

the patch fails due to the ip variable being named as luci.ip import name
```
local ip = require "luci.ip"
[...]
local ip, scope = url.host:match("^([^%%]+)%%(.+)$")
url.ip6addr = ip.IPv6(ip or url.host)
```

and the direction being used here is url.ip6addr (without %br-lan) so I think that %br-lan should be included in url.ip6addr, no?
```
local addr = tostring(url.ip6addr or url.ipaddr or url.host)
local sock, code, msg = nixio.connect(addr, url.port)
```

jow-, 2019-12-09:
The two referenced commits should solve this issue, please give it a try and let me know if the fix is confirmed so that I can cherry pick it to 19.07.

ilario, 2019-12-10:
Many thanks @jow- ! 
I tested and it works :D

jow-, 2019-12-10:
Great - thank you for confirming.

closed 2019-12-10 by jow-