openwrt/luci #3381: Is ucitrack useless for procd init?

ysc3839, 2019-12-02:
It seems that `ucitrack` is for non-procd init.
https://github.com/openwrt/luci/blob/a67c1c4769349b283d1589c53ecd2c29253a9074/modules/luci-base/root/etc/init.d/ucitrack#L12
However, some packages, which use procd init, still define a `ucitrack` config.
https://github.com/openwrt/packages/blob/2dd76ae7658f7a1c44e2626f64d7e945982192d1/net/nextdns/files/nextdns.init#L74-L76
https://github.com/openwrt/luci/blob/a67c1c4769349b283d1589c53ecd2c29253a9074/applications/luci-app-nextdns/root/etc/uci-defaults/60_luci-nextdns#L3-L8

jow-, 2019-12-02:
It indeed is useless for proxy-enabled packages/unit scripts.

ysc3839, 2019-12-03:
@jow- So should we remove these definitions?

jow-, 2019-12-03:
If the underlying service is procd enabled then yes.

feckert, 2019-12-03:
> If the underlying service is procd enabled then yes.

This only applies if a _service_tiggers_ is defined and set in a procd init script.

closed 2019-12-03 by jow-

ysc3839, 2019-12-03:
@jow- @feckert Let me create a pull request?

feckert, 2019-12-03:
I think where it is possible we should remove the change in the ucitrack in small steps.
Service per Service.