openwrt/luci #3384: Improper automatic editing of LAN interface

Faerbit, 2019-12-02:
Setup:
Device: TP-Link WDR 3600 v1
 Versions: OpenWrt 19.07.0-rc2 r10775-db8345d8e4 / LuCI openwrt-19.07 branch git-19.334.63023-039ef1f
LAN Interface with VLAN 1 with ports 0t 2 3
WAN Interface with VLAN 2 with ports 0t 1
SERVER Interface with VLAN 3 with ports 0t 4 5
Action:
I decided I did not need the SERVER interface and VLAN 3 anymore. Therefore I wanted to switch ports 4 and 5 to VLAN 1.  I made this change in LuCi using "Save and Apply" after input of my changes.
Result:
 * Automatic edit of my LAN interface from physical interface `eth0.1` to `eth0`. At least that was communicated to me via a status message
 * Unable to use ethernet for internet nor router access (WiFi still worked)

Expected Result:
Leave my LAN interface alone, especially if it breaks things.

jow-, 2019-12-02:
If you modify vlans it is expected that you know what you do.

closed 2019-12-02 by jow-

Faerbit, 2019-12-06:
I'm not sure I follow.
IMHO the bug is not that stuff stops working, if it is improperly configured.
The bug is that I observed an automatic configuration change, which resulted in an improper configuration.

jow-, 2019-12-06:
I cannot reproduce the problem. You need to be more specific in what you did exactly. FOr example you cannot assign ports 4 and 5 to vlan 1 since untagged in multiple vlans is denied. So did you delete the vlan 3 first, then reassign the ports, or did you first assign the ports to vlan 1, then set them to disabled in vlan 3, then deleted vlan 3 etc.?

Faerbit, 2019-12-06:
Initial situation:
[image]
Changing LAN3 and 4 to VLAN 1 and then clicking save:
[image]

I just noticed, that I can further isolate this:
From off:
[image]
To untagged:
[image]

But only for Port "LAN 4"/ Port 5! If I try this with Port "LAN 3"/Port 4 this does not happen.

jow-, 2019-12-06:
Still cannot reproduce it, my test device here behaves differently. Please provide the complete `/etc/config/network`, as well as the complete output of `ubus call network.interface dump`

Faerbit, 2019-12-06:
I redacted some information due to privacy concerns. I hope this is still helpful:
https://gist.github.com/Faerbit/a4b15b08f445b0d501b6fa376e723017

jow-, 2019-12-06:
Thanks. Unfortunately I need the initial state of the configuration leading to the error. I also require the contents of ´/etc/board.json`.

Faerbit, 2019-12-06:
I noticed a discrepancy between `/etc/board.json` and `/etc/config/network`. Though even a reboot did not clear that up. Is this my error?
https://gist.github.com/Faerbit/e056fc8e3233f826838f22a93ef76ed5